actuallymentor/battery #90: installation without homebrew and xcode

MKaiser96, 2023-02-25:
Is it possible to use this tool without the installation of homebrew and xcode? These additional tools need a lot of free disk space. I don't like to install this tools.

cowpod, 2023-02-26:
There's AlDente, but if you want the same features as this you'll need to pay ~$10 for AlDente Pro.
https://apphousekitchen.com

jakjakob, 2023-03-01:
> Is it possible to use this tool without the installation of homebrew and xcode? These additional tools need a lot of free disk space. I don't like to install these tools.

There is a version in beta here
https://github.com/actuallymentor/battery/releases

MKaiser96, 2023-03-04:
The installation worked fine. The only problem I have is that I use my MacBook Pro M1 with an external display. When I activate your tool and the display of my MacBook is closed, the MacBook goes to sleep mode. When I open the display it works fine, but generally the display is closed.

actuallymentor, 2023-03-07:
> When I activate your tool and the display of my MacBook is closed, the MacBook goes to sleep mode. When I open the display it works fine, but generally the display is closed.

This is a known issue in #28, I've not been able to reproduce it.