actuallymentor/battery #91: Add charging indicator to icons when charging is allowed

WardsParadox, 2023-02-27:
**Is your feature request related to a problem? Please describe.**
N/A

**Describe the solution you'd like**
I would love battery to have a charging icon. This would let me replace the standard OS battery menubar icon entirely with just battery.

**Describe alternatives you've considered**
Atm I'm using bartender to show the built in icon when the device is charging. When it hits the 80% limit, the icon tends to pop in and out as the device trickle charges through out the day.

**Additional context**
[image]

actuallymentor, 2023-03-07:
I like the idea but it's not high prio, feel free to pull-request it. Otherwise it will take me a while probably.

ballo, 2024-01-26:
Having a % indicator would also be nice. Apple's battery indicator does this in the menu, or optionally in the menu bar icon.