actuallymentor/battery #93: Permission error on startup

EzeGmnz, 2023-02-28:
**What is the issue? (required)**
After opening the battery app it shows the following icon
[image]
it installs the required components successfully and after ~5secs I get the following error
[image]

and the battery icons remains greyed out 
[image]

**Error logs**
`🎉 Battery tool installed. Type "battery help" for instructions.gimenez

Executing nohup curl "https://unidentifiedanalytics.web.app/touch/?namespace=battery" > /dev/null 2>&1
App initialisation process complete
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
Error getting battery status: 
/usr/local/bin/battery: line 149: /usr/local/bin/smc: Permission denied
/usr/local/bin/battery: line 158: /usr/local/bin/smc: Permission denied

Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Refreshing tray icon...
enable_battery_limiter exec complete
Error getting battery status: 
/usr/local/bin/battery: line 149: /usr/local/bin/smc: Permission denied
/usr/local/bin/battery: line 158: /usr/local/bin/smc: Permission denied

Error generating menu: 
TypeError: Cannot destructure property 'battery_state' of '(intermediate value)' as it is undefined.`

ioExpander, 2023-03-01:
Hi. I'm getting the same error on an M1 Macbook Air.

actuallymentor, 2023-03-07:
I think this should be fixed now, can you try to run `battery uninstall` in the terminal and then run the GUI app again?

thomasboleyn, 2023-03-08:
> I think this should be fixed now, can you try to run `battery uninstall` in the terminal and then run the GUI app again?

Thanks, works like a charm now.

EzeGmnz, 2023-03-15:
> I think this should be fixed now, can you try to run `battery uninstall` in the terminal and then run the GUI app again?

Worked, thanks!

closed 2023-05-29 by EzeGmnz