actuallymentor/battery #95: Separate brew package for CLI script and Electron GUI tool

ioExpander, 2023-03-02:
**Is your feature request related to a problem? Please describe.**
As I find Electron apps to be a little bloated, I'm trying to avoid using the battery GUI. I'm currently using the CLI with a quick shell script plugin build for the [xbar app](https://github.com/matryer/xbar)
Unfortunately `brew install battery` installs both the CLI and the GUI.

**Describe the solution you'd like**
Having a separate package to install the CLI standalone eg : `brew instal battery-cli`

**Other ideas considered**
Build a GUI using [Tauri](https://github.com/tauri-apps/tauri) - but I'm trying to KISS. The xbar plugin only takes 15 lines of bash script

**Additional context**
[Plugin Code Available Here](https://gist.github.com/ioExpander/8f578cccd8d7346190a11fb9a1d7fe71)

Zeroscl, 2023-03-06:
I have been investigating what you say about the xbar app and it seems interesting, but is the GUI that is installed with the code that you share us functional with the Battery CLI? 

Or is it just to see the app icon on the bar? 

Thanks for a great contribution.

Regards

Pablo

Zeroscl, 2023-03-06:
> I have been investigating what you say about the xbar app and it seems interesting, but is the GUI that is installed with the code that you share us functional with the Battery CLI?
> 
> Or is it just to see the app icon on the bar?
> 
> Thanks for a great contribution.
> 
> Regards
> 
> Pablo

I tried installing xbar since it seemed like a great app, but strangely nothing worked, from the Cardano price monitor to the CPU temperature monitor.

I imagine it may be a compatibility issue with the MB Pro M1, I even tried with Battery installed but not running the GUI and your code didn't work for me either. Something you're not seeing? that you can guide me?

Thank you 

Regards

ioExpander, 2023-03-07:
Hi @Zeroscl. Thank you for showing interest. This issue is more about creating a brew package for the battery CLI rather than discussing xbar. I just wanted to provide some context. I've added a screenshot [here](https://gist.github.com/ioExpander/8f578cccd8d7346190a11fb9a1d7fe71) to help you. Can we continue that conversation on the gist page so that we keep the focus on creating a brew package for this issue ?

actuallymentor, 2023-03-07:
@ioExpander you're purely asking for a brew package of the CLI?

ioExpander, 2023-03-07:
yes 👍

actuallymentor, 2023-03-07:
For me the current CLI is plenty so I won't make it, but feel free to make a cask, I'm not against in principle.

kevinm6, 2023-03-08:
@actuallymentor I have the same idea about Electron, but I'm thinking about the best solution.
I think the implementation of @ioExpander using **xbar** or **swiftbar** is specific to him.

I can create a PR to `homebrew-cask` for a new `battery-cli` and maybe you can replace in the README the method to install only **cli**.

Tell me if you are fine with this or you prefer something different.

kevinm6, 2023-03-09:
Another question @actuallymentor: are there some design motivation for use `/usr/local/bin` as **installation path** instead of homebrew default `/opt/homebrew` on Apple Silicon macs?

actuallymentor, 2023-03-09:
@kevinm6 the tool was designed before I added it to brew, I was just looking for a logical place for binaries and called `which` on something to see where some common tools were.

I'm more a full stack web/crypto oriented dev, and finding the answer to "what is the right place for binary files" just brought up some stack overflow feuds.

If you have a suggestion to change it, I'm all ears. Does the brew path always exist on macbooks by default?

kevinm6, 2023-03-09:
> When you install Homebrew, as the [docs](https://docs.brew.sh/Installation) says, the  
> `brew —-prefix` is created and added to env **PATH**, based on the *architecture* of the pc (`/usr/local` on Intel Macs, `/opt/homebrew` on Apple Silicon Macs) and after that brew create a symlink of the binary (casks and formulae) to the relative path for binaries.
> This design from brew is used to avoid use `sudo` if is not **really required**.

Thanks for the reply, @actuallymentor.
Since I was trying to make a `battery-cli` [only version](https://github.com/kevinm6/battery-cli) I dive deeper in the code and probably **cli-only** version can diff more than I was thinking.

For the `battery` cask I think your choice are fine, since the output is a `Battery.app`, that installed via **brew** in `/Applications/Battery.app` and symlinked to `/opt/homebrew/Caskroom`, but some problems/conflicts could happens with the CLI version from your [README](https://github.com/actuallymentor/battery#-command-line-version) if both are installed.

closed 2023-07-20 by actuallymentor

melyux, 2023-11-28:
So was a cli-only version added to Homebrew?

kevinm6, 2023-11-28:
> So was a cli-only version added to Homebrew?

In part.
I've [created one](https://github.com/kevinm6/battery-cli), but due to **Homebrew docs** I've avoided to pull a request to add an only-cli version of a cask.

You can install it but it's a **Homebrew tap**, so it requires an extra step that's reported in the **README**.