actuallymentor/battery #98: Problems updating to version 1.1.1

Zeroscl, 2023-03-06:
Unfortunately, Battery it was automatically updated from version 1.0.8, which is the last stable one that worked well for me. 

It was updated to 1.1.1, the installation without problems, only when restarting the computer and trying to load the app again (or trying to leave it at the start of the boot) it sends the following error and there was no case, it didn't work anymore.

[image]

Any ideas why it doesn't work? I must say that once it was updated by Brew, I uninstalled it and tried to update it by the .DMG of the page.

(Continuing with previous threads I did not insist with version 1.0.9 or the following ones because they did not work for me either and I always returned to 1.0.8)

Thanks

Pablo

Zeroscl, 2023-03-07:
> Unfortunately, Battery it was automatically updated from version 1.0.8, which is the last stable one that worked well for me.
>
> It was updated to 1.1.1, the installation without problems, only when [...]

After triying several things, uninstall everything, clean and version 1.1.1 works only when you install for the first time and run the script initially.

[image]

If later, you try to restart the M1 either with Battery in the boot sona to start the login or you execute it after having restarted everything, it sends the same error and you must start from scratch, uninstall everything, delete the script and reinstall.

Regards

Pablo

[Last 1000 lines logs files.pdf](https://github.com/actuallymentor/battery/files/10904102/Last.1000.lines.logs.files.pdf)

actuallymentor, 2023-03-07:
Hmm very strange, I've not been able to reproduce this. Would you show me the output of:

```
which battery
which smc
```

kevinm6, 2023-03-09:
> Any ideas why it doesn't work? I must say that once it was updated by Brew, I uninstalled it and tried to update it by the .DMG of the page

@Zeroscl why you did this after it was updated with **brew**?

jakjakob, 2023-03-13:
I had the same problem, but after trying again the next day it worked without problems (after quitting obviously). Just typing `battery status` said to me that it hadn't permissions (maybe that can help with the fix, don't remember what exactly it said)

jakjakob, 2023-03-13:
#93 seems to have the same problem maybe?

actuallymentor, 2023-07-20:
Can you confirm this is still an issue in v1.1.5?