actuallymentor/battery #100: External display not working when charge limit is enabled

n3w6i3, 2023-03-08:
Since updating the app yesterday I noticed that my external display (connected via usb-c thunderbolt 3) is not working when my mac is in clamshell mode and i have charging limited to default 80% (and battery is full). Worth noting that I am using GUI and not CLI, but since this issue is persisting I tried playing with CLI commands and no luck in solving the problem. 

It makes sense that if the battery is not charing and if a macbook is closed that the display is off, but I had no problems using the app this way till now and would be very grateful to getting a workaround as I prefer just one cable connected to my mac. 

Kind regards

gekoxyz, 2023-03-11:
I have the same problem, except i have the external display connected via HDMI. I have 86% battery but the status is "Power Source: Battery". I'm also using the GUI

sunneydev, 2023-03-11:
Once it discharges it works just fine.

actuallymentor, 2023-03-13:
Clamshell users, please try [this beta release](https://github.com/actuallymentor/battery/releases/tag/v1.1.4-beta) and thumbs up/down it to let me know if it works for you now!

There is now a new GUI option:

[image]

n3w6i3, 2023-03-13:
Just tried the update, unfortunately does not work for me in clamshel,  even when force discharging is enabled and battery is full.

actuallymentor, 2023-03-13:
@n3w6i3 you're supposed to have it disabled :)

actuallymentor, 2023-03-14:
Closed by release [v1.1.4](https://github.com/actuallymentor/battery/releases/tag/v1.1.4). Please reopen if you have the same issue with that release (with debug info).

closed 2023-03-14 by actuallymentor