actuallymentor/battery #103: Is there a way to enable charging off on login / startup?

anarchy89, 2023-03-08:
Is there a way to run the command `battery charging off` on every startup?

jakjakob, 2023-03-13:
Maybe [this thread](https://superuser.com/questions/229773/run-command-on-startup-login-mac-os-x) can help you

anarchy89, 2023-03-14:
> Maybe [this thread](https://superuser.com/questions/229773/run-command-on-startup-login-mac-os-x) can help you 
> 
> 

So what's the syntax for launchd for battery charging off?

kevinm6, 2023-03-14:
> > Maybe [this thread](https://superuser.com/questions/229773/run-command-on-startup-login-mac-os-x) can help you
> 
> So what's the syntax for launchd for battery charging off?

This should work, put this in `~/Library/LaunchAgents` as `<name>.plist`:
```xml
<?xml version=\"1.0\" encoding=\"UTF-8\"?>
<!DOCTYPE plist PUBLIC \"-//Apple//DTD PLIST 1.0//EN\" \"http://www.apple.com/DTDs/PropertyList-1.0.dtd\">
<plist version=\"1.0\">
  <dict>
    <key>Label</key>
    <string>com.battery_chargin_off</string>
    <key>ProgramArguments</key>
    <array>
      <string>/usr/local/bin/battery</string>
      <string>charging</string>
      <string>off</string>
    </array>
    <key>RunAtLoad</key>   <!— This run the above array at startup -->
    <true/>
  </dict>
</plist>
```

then save it and run in a terminal: `sudo launchctl load ~/Library/LaunchAgents/<name.plist>`

kevinm6, 2023-03-15:
It should be possible!
You can create a `shell script` that call Apple script with something like `osascript -e "display dialog "` and on the output it runs or not `battery charging off`!

After you create the script you have to change the <Array> `ProgramArguments` in the `.plist` file

anarchy89, 2023-03-15:
K> It should be possible!
> You can create a `shell script` that call Apple script with something like `osascript -e "display dialog "` and on the output it runs or not `battery charging off`!
> 
> After you create the script you have to change the <Array> `ProgramArguments` in the `.plist` file

So like,

```

osascript - "<<EOF 
on run
  tell application "Terminal"
    do script "battery charging off"
  end tell
end run
EOF
```

I save it as batteryoff.scpt

Then I the plist file I do 

`<string>/path/to/batteryoff.scpt</string>`?

kevinm6, 2023-03-15:
Ehm, no! You can do something in that way, personally I prefer use shell script more and leave Apple script do only the **dialog** part.

Something like:

```bash
#!/usr/bin/env bash

output=$(osascript -e 'tell app "System Events" to display dialog “Set battery charging off?”')

if [[ $out == “button returned:OK” ]]; then
  battery charging off
fi
```

- save it somewhere as `<name>` (_the `.sh` or other extensions part is not mandatory on Unix_),
- make it **executable** with `chmod +x <name>` 

Now in the `.plist` change to:

```xml
<?xml version=\"1.0\" encoding=\"UTF-8\"?>

<!DOCTYPE plist PUBLIC \"-//Apple//DTD PLIST 1.0//EN\" \"http://www.apple.com/DTDs/PropertyList-1.0.dtd\">
<plist version=\"1.0\">
  <dict>
    <key>Label</key>
    <string>com.battery_charging_off</string>
    <key>ProgramArguments</key>
    <array>
      <string>full_path_to_file_above(name)></string>
    </array>
    <key>RunAtLoad</key>
    <true/>
  </dict>
</plist>
```

Probably there’s some fix to do, but this is the general idea!
When the system startup, it should prompt you the dialog and in case of “OK” pressed, it should run `battery charging off`.

actuallymentor, 2023-07-20:
This is a good discussion, though outside the scope of the app. I'll close this issue for my own admin, but feel free to continue discussing!

closed 2023-07-20 by actuallymentor