actuallymentor/battery #105: Percentage icon not accurate

earthsound, 2023-03-11:
**What is the issue? (required)**
The icon doesn't match the remaining battery percentage. 

GUI logs show:

```
Display percentage 5 based on 46
```

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Battery is 46% (haven't tested a range of percentage left yet)
2. Battery icon shows a thin sliver like it did when it was close to 0%.

**Expected behavior (required)**
Based on recent changes, I expected the icon to change to reflect the percentage left. 

**Screenshots (optional)**
[image]

**Error logs**
<details><summary>battery logs</summary>
```
battery logs
👾 Battery CLI logs:

03/10/23-22:38:19 - Battery at 81% (target 75%)
03/10/23-22:39:19 - Battery at 81% (target 75%)
03/10/23-22:39:19 - Battery at 81% (target 75%)
03/10/23-22:40:19 - Battery at 81% (target 75%)
03/10/23-22:40:19 - Battery at 81% (target 75%)
03/10/23-22:41:19 - Battery at 80% (target 75%)
03/10/23-22:41:19 - Battery at 80% (target 75%)
03/10/23-22:42:19 - Battery at 80% (target 75%)
03/10/23-22:42:19 - Battery at 80% (target 75%)
03/10/23-22:43:19 - Battery at 80% (target 75%)
03/10/23-22:43:19 - Battery at 80% (target 75%)
03/10/23-22:44:19 - Battery at 80% (target 75%)
03/10/23-22:44:19 - Battery at 80% (target 75%)
03/10/23-22:45:19 - Battery at 80% (target 75%)
03/10/23-22:45:19 - Battery at 80% (target 75%)
03/10/23-22:46:19 - Battery at 80% (target 75%)
03/10/23-22:46:19 - Battery at 80% (target 75%)
03/10/23-22:47:19 - Battery at 80% (target 75%)
03/10/23-22:47:20 - Battery at 80% (target 75%)
03/10/23-22:48:19 - Battery at 80% (target 75%)
03/10/23-22:48:20 - Battery at 80% (target 75%)
03/10/23-22:49:19 - Battery at 80% (target 75%)
03/10/23-22:49:20 - Battery at 80% (target 75%)
03/10/23-22:50:19 - Battery at 80% (target 75%)
03/10/23-22:50:20 - Battery at 80% (target 75%)
03/10/23-22:51:19 - Battery at 79% (target 75%)
03/10/23-22:51:20 - Battery at 79% (target 75%)
03/10/23-22:52:19 - Battery at 79% (target 75%)
03/10/23-22:52:20 - Battery at 79% (target 75%)
03/10/23-22:53:19 - Battery at 79% (target 75%)
03/10/23-22:53:20 - Battery at 79% (target 75%)
03/10/23-22:54:19 - Battery at 79% (target 75%)
03/10/23-22:54:20 - Battery at 79% (target 75%)
03/10/23-22:55:19 - Battery at 79% (target 75%)
03/10/23-22:55:20 - Battery at 79% (target 75%)
03/10/23-22:56:19 - Battery at 78% (target 75%)
03/10/23-22:56:20 - Battery at 78% (target 75%)
03/10/23-22:57:20 - Battery at 78% (target 75%)
03/10/23-22:57:20 - Battery at 78% (target 75%)
03/10/23-22:58:20 - Battery at 78% (target 75%)
03/10/23-22:58:20 - Battery at 78% (target 75%)
03/10/23-22:59:20 - Battery at 78% (target 75%)
03/10/23-22:59:20 - Battery at 78% (target 75%)
03/10/23-23:00:20 - Battery at 78% (target 75%)
[... 151 lines cut ...]
Resetting interface timer speed
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Get active logo for 46
Refreshing interface update timer
Display percentage 5 based on 46
Setting interface refresh speed to 10 minutes
Battery status: 
{"percentage":"46","remaining":"20:00","charging":true,"discharging":false,"maintain_percentage":"75","battery_state":"46% (20:00 remaining)","daemon_state":"smc charging enabled"}

📁 Config folder details:

total 8760
drwxr-xr-x   6 earthsound  staff   192B Mar  7 22:46 .
drwxr-xr-x+ 62 earthsound  staff   1.9K Mar  7 22:46 ..
-rwxr-xr-x   1 earthsound  staff   166K Mar 11 14:24 battery.log
-rw-r--r--   1 earthsound  staff     6B Mar 10 22:14 battery.pid
-rw-r--r--   1 earthsound  staff   3.1M Mar 11 14:21 gui.log
-rw-r--r--   1 earthsound  staff     3B Mar 10 22:14 maintain.percentage

⚙️  Battery data:

03/11/23-14:24:11 - Battery at 46% (20:00 remaining), smc charging enabled
03/11/23-14:24:11 - Your battery is currently being maintained at 75%
```
</details>

jakjakob, 2023-03-13:
I have that problem too. Also at 79% it shows the 50%, etc.

jonahclarsen, 2023-03-14:
Having this problem too. Seems like it was fine at/above 50% but as soon as it hit 49%, it looks like it's at ~5%.

bonjourlemonde, 2023-03-17:
same here, unfortunately.,

igorkulman, 2023-03-23:
I have the same problem, looking at the code in https://github.com/actuallymentor/battery/blob/848ee9ba175b92e6a75acac8d60e9a956bc7a88f/app/modules/theme.js#L26 the sorted percentages are just wrong

```javascript
[0, 10, 100, 15, 20, 25, 30, 35, 40, 45, 5, 50, 55, 60, 65, 70, 75, 80, 85, 90, 95]
```

typical JavaScript 🤷‍♀️

igorkulman, 2023-03-23:
Here is the fix https://github.com/actuallymentor/battery/pull/119 with this the code you get the correct `"Display percentage 45 based on 46"`

PR #119 (2023-03-26): Correct way to determine battery percentage icon for the menu tray
As mentioned in #105 the way the battery percentage for the menu tray icon is computed is not correct, the issue mentions `Display percentage 5 based on 46`. The reason for this is that in https://github.com/actuallymentor/battery/blob/848ee9ba175b92e6a75acac8d60e9a956bc7a88f/app/modules/theme.js#L26 the sort is lexicographical resulting in `[0, 10, 100, 15, 20, 25, 30, 35, 40, 45, 5, 50, 55, 60, 65, 70, 75, 80, 85, 90, 95]`

This PR fixes #105 using the correct sort and also simplifying the way the percentage is  fetched from  [... cut]
closed 2023-03-26 by actuallymentor