actuallymentor/battery #107: It appears that macos is stopping the charge at 80%

jpjenkins, 2023-03-13:
I am attempting to write a script that mimics the aldente program feature of calibrating a battery in which it discharges to 15%, charges to 100% and holds for 1 hour, then discharges to 80% and maintains the charge. 

My script is [here](https://github.com/jpjenkins/actuallymentor-batteryCalibrationScript) if you'd like to take a look 

My issue is that after the discharge, upon recharging it stops at 80% and appears to override 'battery' commands completely. Do you have any suggestions? Thanks

vitork, 2023-03-22:
Since I'm using a third party baterry manager, I disabled the "Optimized Battery Charging" option from the Battery options.
[image]

actuallymentor, 2023-07-20:
It's on my radar, closing this issue as duplicate. Thanks for reporting.

closed 2023-07-20 by actuallymentor