actuallymentor/battery #109: Top bar icon disapperas

Vultilion, 2023-03-14:
Whenever the % next to the original battery icon is activated, the icon of the battery-limiter is not visible.

earthsound, 2023-03-14:
@Vultilion do you have a MBP with a notch? If so, it may be hidden due to too many icons in the menu bar. If you have enough room it should show. 
[image]

There are a few options:

- Close apps that have icons in the menu bar
- Hide the menu bar icon for apps that allow it
- Use a tool (e.g., [Hidden Bar](https://github.com/dwarvesf/hidden), [iBar](https://apps.apple.com/us/app/ibar-menubar-icon-control-tool/id6443843900), and [Bartender](https://www.macbartender.com/)) that provides more options for autohiding/showing menu bar icons.

actuallymentor, 2023-07-20:
Thanks for responding to this @earthsound

closed 2023-07-20 by actuallymentor