actuallymentor/battery #111: When allow force discharing is enabled, and disabled, the battery limit resets to 80%

HeinKeyser10, 2023-03-15:
**What is the issue? (required)**
A clear and concise description of what the bug is.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior (required)**
A clear and concise description of what you expected to happen.

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem. On a Mac you can click `control`+`command`+`shift`+`4` to create a screenshot that you can paste into this field with `command`+`4`.

**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.

**Additional context (optional)**
Add any other context about the problem here.

jakjakob, 2023-03-17:
similar issue to #108

HeinKeyser10, 2023-03-17:
Duplicate

HeinKeyser10, 2023-03-17:
Duplicate

closed 2023-03-17 by HeinKeyser10