actuallymentor/battery #116: Is there a way to stop my phone charging while connected to my macbook through type c cable?

Naim995, 2023-03-16:
Perhaps this is not a question for here, but I will make it anyway. Maybe this would serve as a suggestion to add this feature in the future.
I am software engineer and I run tests on my mobile devices. However, I don't want my mobile devices to suck the battery out of my macbook so I was wondering if turning this off was even possible ?

I heard there are some data only cables but I didn't have any luck finding one

Anatharias, 2023-03-24:
This would be awesome, but would require the same kind of SMC driver on the iPhone to prevent the charge.

amazonsingh, 2023-05-27:
maybe connect your iphone to xcode using wifi debugging

actuallymentor, 2023-07-20:
Unknown. Let me know if you figure out how to do this.

closed 2023-07-20 by actuallymentor