actuallymentor/battery #121: Resubmission: App installation causes noticeable connection delay between MBP M1 and Mac Studio Display

lyubomirg1995, 2023-03-26:
**What is the issue? (required)**
Following app installation, a connection delay (screen and audio) of 10-14 seconds occurs between MacBook Pro M1 and Mac Studio Display.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Click battery icon on menu bar
2. Select "Advanced Settings"
3. Click "Allow force-discharging"
4. Click confirm
5. Open Terminal
6.  type "battery on", press enter.
7. Unplug external display from laptop
8. "battery off" then press enter
9. Play a video on YouTube with audio on
10. Connect cable to external display (Mac Studio)
11. Observe 3-4 second delay of external display turning on
12. Observe another 10 seconds before audio begins playing on external display

**Expected behavior (required)**
Laptop audio is immediately interrupted after plugging in laptop to external display, then audio plays on external display 1-2 seconds later. 

**Screenshots (optional)**
N/A

**Error logs**
03/26/23-01:32:24 - Battery at 90% (target 80%)
03/26/23-01:33:24 - Battery at 90% (target 80%)
03/26/23-01:34:24 - Battery at 90% (target 80%)
03/26/23-01:35:24 - Battery at 90% (target 80%)
03/26/23-01:36:24 - Battery at 90% (target 80%)
03/26/23-01:37:24 - Battery at 90% (target 80%)
03/26/23-01:38:24 - Battery at 90% (target 80%)
03/26/23-01:39:24 - Battery at 89% (target 80%)
03/26/23-01:40:24 - Battery at 89% (target 80%)
03/26/23-01:41:24 - Battery at 89% (target 80%)
03/26/23-01:42:24 - Battery at 89% (target 80%)
03/26/23-01:43:24 - Battery at 89% (target 80%)
03/26/23-01:44:24 - Battery at 89% (target 80%)
03/26/23-01:45:24 - Battery at 88% (target 80%)
03/26/23-01:46:24 - Battery at 88% (target 80%)
03/26/23-01:47:25 - Battery at 88% (target 80%)
03/26/23-01:48:25 - Battery at 88% (target 80%)
03/26/23-01:49:25 - Battery at 87% (target 80%)
03/26/23-01:50:25 - Battery at 87% (target 80%)
03/26/23-01:51:25 - Battery at 87% (target 80%)
03/26/23-01:52:25 - Battery at 87% (target 80%)
03/26/23-01:53:25 - Battery at 86% (target 80%)
03/26/23-01:54:25 - Battery at 86% (target 80%)
03/26/23-01:55:25 - Battery at 86% (target 80%)
03/26/23-01:56:25 - Battery at 86% (target 80%)
03/26/23-01:57:25 - Battery at 85% (target 80%)
03/26/23-01:58:25 - Battery at 85% (target 80%)
03/26/23-01:59:25 - Battery at 85% (target 80%)
03/26/23-02:00:25 - Battery at 85% (target 80%)
03/26/23-02:01:25 - Battery at 84% (target 80%)
03/26/23-02:02:25 - Battery at 84% (target 80%)
03/26/23-02:03:25 - Battery at 84% (target 80%)
03/26/23-02:04:25 - Battery at 84% (target 80%)
03/26/23-02:05:25 - Battery at 83% (target 80%)
03/26/23-02:06:25 - Battery at 83% (target 80%)
03/26/23-02:07:25 - Battery at 83% (target 80%)
03/26/23-02:08:26 - Battery at 82% (target 80%)
03/26/23-02:09:26 - Battery at 82% (target 80%)
03/26/23-02:10:26 - Battery at 82% (target 80%)
03/26/23-02:11:26 - Battery at 82% (target 80%)
03/26/23-02:12:26 - Battery at 81% (target 80%)
03/26/23-02:13:26 - Battery at 81% (target 80%)
03/26/23-02:14:26 - Battery at 81% (target 80%)
03/26/23-02:15:26 - Battery at 81% (target 80%)
03/26/23-02:16:26 - 🔼🪫 Disabling battery discharging
[... 153 lines cut ...]
Refreshing interface update timer
Battery status: 
{"percentage":"80","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"80% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 10 minutes
Force discharge setting: boolean false

📁 Config folder details:

total 592
drwxr-xr-x@  6 lyubomirgizdarski  staff   192B Mar 26 12:25 .
drwxr-x---+ 54 lyubomirgizdarski  staff   1.7K Mar 26 00:18 ..
-rwxr-xr-x@  1 lyubomirgizdarski  staff    15K Mar 26 12:29 battery.log
-rw-r--r--@  1 lyubomirgizdarski  staff     6B Mar 26 12:25 battery.pid
-rw-r--r--@  1 lyubomirgizdarski  staff   213K Mar 26 12:25 gui.log
-rw-r--r--@  1 lyubomirgizdarski  staff     3B Mar 26 12:25 maintain.percentage

⚙️  Battery data:

03/26/23-12:29:25 - Battery at 80% (attached; remaining), smc charging disabled
03/26/23-12:29:25 - Your battery is currently being maintained at 80%
Battery CLI utility v1.0.10
Lyubomirs-MBP:~ lyubomirgizdarski$ 

**Additional context (optional)**
During the connection delay, laptop screen displays first on external display, then audio plays for 3-4 seconds on laptop before it is interrupted and is played on external display.

lyubomirg1995, 2023-03-28:
Uninstalling the application does not remove the issue.

actuallymentor, 2023-07-20:
Thanks for resubmitting @lyubomirg1995.

I'm afraid I have no way to reproduce or test this since I do not have such a display. What I can say is that when `battery` is uninstalled, it undoes all changes to the system. If you think it did not, you can reset your SMC (which is where the changes are made, and I'm 99% sure also reversed).