actuallymentor/battery #122: visudo install step fails during CLI install: cp /private/etc/sudoers.d/battery: No such file or directory

GeorgFleig, 2023-03-31:
Thank you for creating this awesome tool and saving our batteries' lives!

**What is the issue? (required)**
visudo install step fails during CLI install and later with `battery visudo`.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Run CLI installer as described in the readme
2. Enter password

**Expected behavior (required)**
Expected a clean install without errors.

**Error logs**
```
curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash                                                                           

####################################################################
# 👋 Welcome, this is the setup script for the battery CLI tool.
# Note: this script will ask for your password once or multiple times.
####################################################################

Password:
🔋 Starting battery installation
[ 1 ] Superuser permissions acquired.
[ 2 ] Downloading latest version of battery CLI
[ 3 ] Move smc to executable folder
[ 4 ] Writing script to /usr/local/bin/battery for user georg
[ 5 ] Setting correct file permissions
[ 6 ] Setting up visudo declarations
cp: /private/etc/sudoers.d/battery: No such file or directory
chmod: /private/etc/sudoers.d/battery: No such file or directory
[ 7 ] Removing temp folder /Users/georg/.battery-tmp

🎉 Battery tool installed. Type "battery help" for instructions.
```
Manual invocation after installation:
```
battery visudo 
Password:
cp: /private/etc/sudoers.d/battery: No such file or directory
chmod: /private/etc/sudoers.d/battery: No such file or directory
```

alin-strongdm, 2023-04-06:
Same here, I was able to fix by creating the sudoers.d folder manually:
1. `sudo mkdir -p /private/etc/sudoers.d`
2. `battery reinstall`

alienator88, 2023-04-06:
Same here, I was able to fix by creating the sudoers.d folder manually:
1. `sudo mkdir -p /private/etc/sudoers.d`
2. `battery reinstall`

closed 2023-04-21 by actuallymentor