actuallymentor/battery #128: battery maintain level not persistent across reboots

jhoogeboom, 2023-04-13:
**What is the issue? (required)**
The battery maintain percentage is reset to 80 on reboot

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
Used the CLI with `battery maintain 70`
I reboot
When I open the app it says it will charge up to 80%
It starts charging
When I open `.battery/maintain.percentage` it states `80`

**Expected behavior (required)**
The maintain percentage is persistent across reboots

**Error logs**
```
👾 Battery CLI logs:

04/13/23-09:50:01 - Debug trail. User: jorishoogeboom, config folder: /Users/jorishoogeboom/.battery, logfile: /Users/jorishoogeboom/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
04/13/23-09:50:01 - No setting to recover, exiting
04/13/23-09:50:01 - Not triggering discharge as it is not requested
04/13/23-09:50:01 - Charging to and maintaining at 80% from 79%
04/13/23-09:50:29 - Not triggering discharge as it is not requested
04/13/23-09:50:29 - Charging to and maintaining at 70% from 80%
04/13/23-09:50:29 - Charge above 70
04/13/23-09:50:29 - 🔌🪫 Disabling battery charging
04/13/23-09:51:52 - Debug trail. User: jorishoogeboom, config folder: /Users/jorishoogeboom/.battery, logfile: /Users/jorishoogeboom/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
04/13/23-09:51:53 - No setting to recover, exiting
04/13/23-09:51:54 - Not triggering discharge as it is not requested
04/13/23-09:51:54 - Charging to and maintaining at 80% from 84%
04/13/23-09:51:54 - Charge above 80
04/13/23-09:51:54 - 🔌🪫 Disabling battery charging

🖥️  Battery GUI logs:

04/13/23-09:51:56 - Your battery is currently being maintained at 80%

Display percentage 80 based on 84
Found image: /Applications/Battery.app/Contents/Resources/battery-active-80-Template.png
Get active logo for 84
Refreshing interface update timer
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Generate app menu percentage: 84 (discharge disallowed, limited on)
Force discharge setting: undefined undefined
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Resetting interface timer speed
Refreshing interface update timer
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 10 minutes
Setting interface refresh speed to 10 minutes
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Refreshing tray icon...
Refreshing tray icon...
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status
Limiter status message: 
04/13/23-09:51:58 - Battery at 84% ((no remaining), smc charging disabled
04/13/23-09:51:58 - Your battery is currently being maintained at 80%

Refreshing interface update timer
Resetting interface timer speed
Force discharge setting: undefined undefined
Generate app menu percentage: 84 (discharge disallowed, limited on)
Display percentage 80 based on 84
Found image: /Applications/Battery.app/Contents/Resources/battery-active-80-Template.png
Get active logo for 84
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Limiter status message: 
04/13/23-09:51:58 - Battery at 84% ((no remaining), smc charging disabled
04/13/23-09:51:58 - Your battery is currently being maintained at 80%

Get active logo for 84
Found image: /Applications/Battery.app/Contents/Resources/battery-active-80-Template.png
Generate app menu percentage: 84 (discharge disallowed, limited on)
Refreshing interface update timer
Display percentage 80 based on 84
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Force discharge setting: undefined undefined
Resetting interface timer speed
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 10 minutes
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 10 minutes
[ update-electron-app ] 
update-not-available
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Refreshing tray icon...
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status
Display percentage 80 based on 84
Found image: /Applications/Battery.app/Contents/Resources/battery-active-80-Template.png
Get active logo for 84
Generate app menu percentage: 84 (discharge disallowed, limited on)
Limiter status message: 
04/13/23-09:52:33 - Battery at 84% ((no remaining), smc charging disabled
04/13/23-09:52:33 - Your battery is currently being maintained at 80%

Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Resetting interface timer speed
Refreshing interface update timer
Force discharge setting: undefined undefined
Setting interface refresh speed to 10 minutes
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Refreshing tray icon...
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status
Generate app menu percentage: 84 (discharge disallowed, limited on)
Get active logo for 84
Display percentage 80 based on 84
Force discharge setting: undefined undefined
Limiter status message: 
04/13/23-09:52:33 - Battery at 84% ((no remaining), smc charging disabled
04/13/23-09:52:33 - Your battery is currently being maintained at 80%

Refreshing interface update timer
Resetting interface timer speed
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Found image: /Applications/Battery.app/Contents/Resources/battery-active-80-Template.png
Setting interface refresh speed to 10 minutes
Battery status: 
{"percentage":"84","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"84% (unknown remaining)","daemon_state":"smc charging disabled"}

📁 Config folder details:

total 72
drwxr-xr-x@  6 jorishoogeboom  staff   192B Apr 13 09:51 .
drwxr-xr-x+ 73 jorishoogeboom  staff   2.3K Apr 13 09:49 ..
-rw-r--r--@  1 jorishoogeboom  staff   1.1K Apr 13 09:55 battery.log
-rw-r--r--@  1 jorishoogeboom  staff     4B Apr 13 09:51 battery.pid
-rw-r--r--@  1 jorishoogeboom  staff    20K Apr 13 09:52 gui.log
-rw-r--r--@  1 jorishoogeboom  staff     3B Apr 13 09:51 maintain.percentage

⚙️  Battery data:

04/13/23-09:55:31 - Battery at 84% (attached; remaining), smc charging disabled
04/13/23-09:55:31 - Your battery is currently being maintained at 80%
Battery CLI utility v1.0.11
```

PatiB87, 2023-04-13:
For me it currently is not only not persistent across reboots but changes very often. I set it to e.g. 50% via CLI... 5 min later (while actively using the MacBook the whole time) I notice that it started charging to 80 again...!
Maybe some AppNap issue or something?

ballo, 2023-04-26:
This is the #1 issue for me. I just rebooted and forgot to run the battery command which has to be done every single time manually. It's just a poor package.

RoHe56, 2023-04-26:
I also have the same problem, after reboot it doesn't keep the set parameters. I can't imagine that I will set the charge parameters after every reboot :((

actuallymentor, 2023-07-20:
This should not happen. I'll add it to the list (but am busy so have no timeline). Help and PR welcome.

Hyp3rSoniX, 2023-07-27:
AlDente tells the User to turn off optimized charging in the settings. The setting that automatically sets the charging limit to 80%.

Maybe that's what's causing this issue?

It can be turned off in the Battery settings > click on the (i) at the right of battery condition, and there turn of optimized charging.

Not sure if that really is the cause of the problem, but that just struck out to me when installing AlDente.

closed 2023-08-20 by actuallymentor

ballo, 2024-01-01:
This time it charged my battery 100% (which I didn't ask for it to do) and basically killed my M1 MacBook Pro until I force reset the SMC (10 second power button hold).

I'd tell you the version but the CLI is silly dumb:

`> battery -v
> battery -version
> battery --version`

PR #166 (2023-08-20): CLI GUI compatibility
Fix for issue #149 and #128: The percentage set in the CLI is not persistent upon GUI reboot, and reset to 80%. 

**Problem**: On launch of the GUI, the maintain.percentage file is deleted, causing the GUI to recreate the file with default value of 80.
this is caused by the call "battery maintain stop" which deletes the maintain.percentage file. 

This issue is easily reproduced by changing the `battery maintain 70` via CLI, and then launching the GUI. 
 
**Solution**: Remove the deletion of file maintain.percentage in "battery maintain stop" to keep persistence of any pr [... cut]