actuallymentor/battery #129: Battery Maintain gets confused with previous values.

Paranoid-Dev, 2023-04-14:
**What is the issue? (required)**
After I updated Mac OS to the latest version (Ventura 13.3.1), `battery maintain` isn't working properly. I'm using the cli and using `battery maintain 50`, `battery maintain 70` etc to set the limit. 
Below is the log after I used it today. Today I had it at maintaining 50%. I used it at 70% yesterday. My battery was at 45% when I pluged it in, and it's now at 70% after a couple of hours. 
I ran 
the command `battery maintain 50` multiple times during the duration of charging. 
Here is the log file included. [logs.txt](https://github.com/actuallymentor/battery/files/11229928/batterylogs.txt)

Paranoid-Dev, 2023-04-14:
Maybe multiple daemons are running at once with different battery maintain values?
I was just using the `battery maintain n` command in the terminal

charlie0129, 2023-04-14:
I had this problem when I used battery before. I was on Monterey but I also encountered this, so it shouldn't be associated with system version..

earthsound, 2023-04-18:
Possibly related to #73?

Paranoid-Dev, 2023-04-18:
@earthsound my bad, your original issue looks the same as what I've reported later. I did search for a previous issue but I missed it. 
I'll keep this issue open since the dev hasn't addressed/even made a comment on #73 yet. 
I'll try my hand at patching it after my mid terms end next week.

Paranoid-Dev, 2023-04-18:
So I stopped the Battery daemon by running `battery maintain stop` and then deleting the contents of the battery dir (the dir where battery.pid, battery.log etc reside) and running `battery maintain 50` again. 
I experienced the same behavior again. (see the included png file - at the bottom, it says charging enabled even with the battery charge above the maintain value)

After going through `battery.sh`, I think I found where the issue lies, but will need to do some more testing
[log.png](https://user-images.githubusercontent.com/62641568/232781161-111a5ad5-2ad8-4b81-b23f-26fec21816dc.png)

earthsound, 2023-04-18:
@Paranoid-Dev fwiw, my report was based on an older version of battery and changes since then may have rendered some of my report moot.

Paranoid-Dev, 2023-04-22:
My fix unfortunately still didn't work but the issue is multiple battery background processes running at once with different values - here's a manual fix : if you kill all the processes but one from your terminal it maintains fine without any bugs. 

I might try to make a small utility with c myself if I can't find how to fix it - shell script + managing background processes by saving the pid of a bash script makes it very hard to debug for me.

martinmenge, 2023-05-17:
I had issue with increasing maintain (from 80 to 100) and stuck on old maintain. Previous attempts included a restart (presumably "flushing" deamons). Today, for first time,  I simply opened maintain.percentage in vim and closed it again. I cannot recall what percentage was in the file but I didn't change it. I noticed about an hour later that it charged to 100 as per latest command.

actuallymentor, 2023-07-20:
Can you confirm this is still an issue in v1.1.5?

closed 2023-07-20 by actuallymentor

Paranoid-Dev, 2023-08-24:
oops I was sick for an entire month and I didn't check my inbox til just now. 
I made a wrapper shell script that more or less fixed the issue for me so I haven't seen if the new version changed anything but I'll try without my script and let you know if the issue still persists. 

Here's the wrapper script I made if you're reading through this issue trying to search for a solution:
```bash
# battery.sh
#!/bin/bash
if (($1>1)); then
  if (($1<101)); then
    battery maintain stop
    launchctl stop com.battery.app
    battery maintain $1
    battery status
  else
    echo integer cannot be more than 100
  fi
else
  echo integer cannot be less than 1
fi
```
and you use it with `./battery 80` to maintain 80% etc.