actuallymentor/battery #133: "Error installing battery limiter: undefined" in Ventura when launched from Finder or Spotlight, but from `open` command

nhooey, 2023-04-23:
Invoking the "Battery" GUI app (`/Applications/Battery.app`) when double-clicking from Finder, or invoking from a Spotlight search, both throw the error: `Error installing battery, limiter: undefined`. Running `open /Applications/Battery.app` from a terminal runs the GUI app without issue, and prevents further invocations from Finder or Spotlight from failing while the app continues to run.

**Error message:**
[image]

The error occurs regardless of the permissions of anything in `~/.battery`, and it only happens when launching Battery from Spotlight, when it's not already running.

**System information:**

1. Mac OS: Ventura `13.3.1 (22E261)`
1. Battery installed only ever from Homebrew cask, updated today
1. Nothing has been installed outside of Homebrew, Nix, and the Mac App Store
1. Permissions opened for config directory: `sudo chmod -Rc a+w ~/.battery`

If I run `open /Applications/Battery.app` from a terminal, the Battery app starts and shows its menulet in the Notification Area:

**Battery.app Notification Area menulet:**

[image]

Then further invocations of "Battery" from Finder or Spotlight have no effect (probably because Mac OS doesn't actually do anything after launching a program that is already running).

If I kill the Battery app by clicking "Quit" on its menulet dropdown menu, then all Finder or Spotlight invocations of Battery show the error dialogue shown above in this message, until battery is run with the command: `open /Applications/Battery.app` again.

So there is something wrong specifically with the Spotlight and Finder invocations of Battery.

rayshan, 2023-04-27:
+1 same here with Version 1.1.4

SeargantSaxton, 2023-05-17:
Mine is not even working from the command line.
same error, undefined

matphillips, 2023-05-26:
I had the same problem but found that if I run "battery uninstall" from command line, the GUI installer then worked fine. I figure it manages to install the CLI tools, but fails for some reason to install the GUI. Repeated attempts to install fail because it is partially installed already. 
Removing and reinstalling worked fine for me. Your mileage may vary :)

nhooey, 2023-07-07:
@SeargantSaxton: After upgrading with Homebrew today, my version of Battery is _also_ now failing when launching with this method that worked before:

> 1. Running `open /Applications/Battery.app` from a terminal

nhooey, 2023-07-07:
> @matphillips: Removing and reinstalling worked fine for me. Your mileage may vary :)

Uninstalling and reinstalling the `battery` cask with Homebrew doesn't fix the problem, running the Battery GUI with all 4 methods mentioned in the issue still fail with the same error message.

actuallymentor, 2023-07-20:
This is very strange and I have no idea what the source of this could me... If anyone has ideas, please let me know!

nhooey, 2023-07-20:
Workaround is to just use another equivalent app called “Aldente”. It’s not
all free, but it’s a lot more reliable, and the free version has enough
features.

I looked at the logging code in this app and it’s really lacking some basic
stuff. There’s not much chance that any of these bugs get fixed.

SangTruongTan, 2023-07-22:
It's also experience on MacOS 14.0 Sonoma. There is no workaround can be a remedy.

actuallymentor, 2023-07-23:
@SangTruongTan what is the manipulation and power cycle steps you took? Can it help me find the cause?

SangTruongTan, 2023-07-23:
Hello @actuallymentor,
I've tried to use the GUI, but it was failed. So, I decided to interact with the CLI. During that time, I made some manipulation such as enable/disable charging, maintain battery level. Afterwards, I rebooted my Mac, when it was ready, I tried opening the GUI app and then it worked find (until now).
[image]

Tweoss, 2023-07-26:
I had this issue and determined by running the executable in terminal that I had an error in my `.bashrc`. Removing those lines fixed the error.

lucc70, 2023-07-30:
@Tweoss 

which lines you removed from the .bashrc file plz ?

Tweoss, 2023-07-30:
> @Tweoss
> 
> which lines you removed from the .bashrc file plz ?

@lucc70 Les lignes étaient pour une autre application qui n'était pas dans le PATH. Alors, malheureusement, l'erreur a prévenu l'ouverture du script de Battery. Peut-être vous pouvez commenter toutes les lignes dans votre `.bashrc` et ouvrir battery une fois plus?

rivolity, 2023-09-13:
@lucc70 @Tweoss @SangTruongTan @actuallymentor @nhooey @matphillips @rayshan @SeargantSaxton 

You can fix that by following these steps.
*I had the same probleme yesterday*

- Ensure that the Battery is installed, even if you have the following error message ` Error installing battery limiter: undefined ` just don't uninstall it.
Run this command in terminal
```
which battery
```
You should get this output
```
/usr/local/bin/battery
```
$\color{orange}{Step:2}$
- Run the following command to update the Battery app from the command line.
*this will call the shell script and download the latest script from the git repo and execute the installation via bash.*

```
battery reinstall
```
Press any key to continue and hit enter, this will prompt you to enter the root password, since the shell script need root access to visudo and other privilege.

Here is a screen capture of the running command

[image]

Here is the latest version after the update via the CLI.

[image]

actuallymentor, 2024-01-17:
Thank you @rivolity!

Can someone else confirm this works for them too? It looks like that then the issue is an old version holding things up.

Please reopen is the above does not fix the issue for you.

closed 2024-01-17 by actuallymentor