actuallymentor/battery #134: Battery icon showing lower battery level than it actually is

chrizz, 2023-04-26:
I've noticed that when battery charge drops somewhere below 50%, the menu bar icon actually shows an almost empty battery. This happens suddenly. When battery level is 50-60 or above the icon shows a more accurate charge state.

Using v1.1.4 with 80% charge limit enabled on a Macbook Pro 14 M2 Pro.

[image]

earthsound, 2023-04-27:
This was fixed with GH-119 which was merged 3/26, after the last release.

kristiankruse, 2023-05-04:
Same issue is still on 1.1.4
[image]

actuallymentor, 2023-07-20:
Should be fixed in v1.1.5 (releasing in 5 min), reopen if it persists please :)

closed 2023-07-20 by actuallymentor