actuallymentor/battery #135: Battery charged to 100% despite limit

rexovas, 2023-04-29:
As title suggests

Installed the battery cli yesterday - computer has not been rebooted, though it was with the lid closed and unplugged overnight.  Today I plugged in the computer to charge when the battery was around 49%.  I left home expecting to come back and find it charged to 80%, however it was charged to 100%.  See the below screenshot.

[image]

rexovas, 2023-04-30:
Tried again last night - plugged it in right before heading to sleep.  This morning it was charged only to 85% and reported `smc charging disabled`, but still above 80%.  Not quite sure why that is.

[image]

charlie0129, 2023-05-10:
This is cause by sleeping, see #123 . I did some analysis in that issue. Probably you can check that issue out.

rexovas, 2023-05-10:
Thanks for the reply, though if sleep is causing the issue, why does the computer not charge to 100% despite being plugged in/lid closed overnight? 

Since that first time it charged to 100%, I haven’t had it do that again, but it has charged ~1-5% above the limit I’ve set.

charlie0129, 2023-05-11:
Charging a couple of percent more seems normal to me if you rebooted your computer or did some other things that caused the script to stop. Looking at the logs should make it easier to debug.

rexovas, 2023-05-11:
Where are the logs located for the cli version

charlie0129, 2023-05-11:
> Where are the logs located for the cli version?

You can run `battery logs` to see it.

actuallymentor, 2023-07-20:
Closing as duplicate of #123

closed 2023-07-20 by actuallymentor