actuallymentor/battery #136: FR: Add to Readme - How does the project compare to the macOS's "Optimized battery charging"?

anijatsu, 2023-04-30:
**Is your feature request related to a problem? Please describe.**
A related question appeared in #11, asking whether it'd be a good idea to disable the system-wide "Optimized battery charging" toggle with this app on.

**Describe the solution you'd like**
Providing more information in README.md on the benefits of using this application rather than the included with macOS "Optimized battery charging" option.
That'd probably involve figuring out how it works - answering questions such as e.g.:
- Since the _Optimized battery charging_ setting's description mentions "learning the charging routine", is there any way to view/tune/reset the details the OS has so far gathered?
- After it reaches a satisfying battery charge level and machine is still plugged in, is it going to switch over to using the power adapter as the sole power source, or will it trickle charge the battery - perhaps based on system load or else?

**Describe alternatives you've considered**
- State that the project has the same goal as the OS setting, but has a stricter scope - it never goes up to 100%.
- Its mechanism is already described - starting at the "Please note that the app:" line. For better reading at a glance, it could be moved into a separate section to make it stand out.
- Perhaps the application could also check whether the OS setting is turned on, and suggest toggling it to avoid interference? Otherwise state it in the readme.md as well.

**Additional context**
N/a

actuallymentor, 2023-07-20:
I'd love a PR for this. Do you have the time?

double-thinker, 2023-08-14:
Maybe this is relevant for context: #163 

The main differences I found are:

- There is no manual control with "Optimized battery charging". For example, I had it for a few weeks and no limit was activated by my MacOS.
- The daemon solution can keep an arbitrary limit rather than only 80%.

That being said, I am now inclined to use only the firmware limit because I expect it to be healthier for the battery (less cycles, developed by Apple, etc.) compared to a userspace solution (like AlDente or similar tools). And the "optimized charging" feature, at least for me, is not very useful as it has not triggered any limit yet. Maybe I am too inconsistent for Apple's ML algorithms 😆

ScienceOwl13, 2023-08-22:
Hey!

I'd be willing to work on a update to the `readme.md` file concerning the difference & advantage of battery vs optimized charging -- but I just want to clarify that adding a section to the readme is the only update, and there isn't anything that has to do with changing the optimize battery on/off in relation to if batteries on/off.

Just wanted some clarification. 

:D