actuallymentor/battery #138: Allow little discharge when activating battery limit

Paaf31, 2023-05-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I heard that microcharging is not very good either for battery life.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Could be great to allow the battery to discharge 1-5% under the 80% limit before allowing it to charge again.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

crbon, 2023-05-24:
Agreed. I think preventing micro charging is an important feature to help maintain a good battery life.
It would be great to see this feature, and have the % drop value adjustable.

AlDente does this and calls this feature [Sailing Mode](https://apphousekitchen.com/feature-explanation-sailing-mode/)

praveeno, 2023-05-28:
for now we can manually handle it
charge till 80% and set maintain mode to 70%, so it will discharge 10% before start charging again.
when charge drop to 70%, again charge to 80%, and set maintain mode to 70%.

Kriepke, 2023-06-23:
I think your tip doesn't work. Especially when you set allow force-discharging.

actuallymentor, 2023-07-20:
@Paaf31 do you have docs where I can read about the impact of microcharging?

Either way, when plugged in, the battery does not dis/recharge, it feeds directly off the charger. There will only be discharging if you are using a weak charger.

actuallymentor, 2023-07-21:
> It's true

@Kriepke docs please :)

Kriepke, 2023-07-22:
You mean docs or proof? ;)

actuallymentor, 2023-07-23:
Your comments are not very helpful.

I'm trying to understand how important this is to battery health. [Something like this](https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries).

Closing until someone can help me understand.

closed 2023-07-23 by actuallymentor

Kriepke, 2023-07-23:
While the charging of the MacBook is paused, i.e. the charge level is exactly at the set charge limit, the power supply is primarily used as a power source. However, small discharges of the battery still occur when a lot of power is needed for a short time. Then the battery of the MacBook steps in as a buffer. If the charge level drops below the charge limit due to these short discharges, Battery-app recharges the battery to the desired charge limit. A second effect on why the battery loses charge is time. If a battery is left for weeks without charging or discharging, it will also lose charge over time. Also, in this case, Battery app would detect that the charge level has dropped below the charge limit and would recharge the battery to the selected charge limit. Furthermore, if you unplug your MacBook only for a short time and use only a couple of percent, your Battery-app would charge MacBook again. **Since it is theoretically healthier for the battery to be charged 10% once instead of 1% 10 times.**

Paaf31, 2023-07-24:
> Your comments are not very helpful.
> 
> I'm trying to understand how important this is to battery health. [Something like this](https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries).
> 
> Closing until someone can help me understand.

Sorry, I don't have any proof or docs regarding of why/how it is bad for battery health. I only came across this infos multiple times 😉