actuallymentor/battery #140: Allow full charge once a month

Paaf31, 2023-05-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I heard it is good practice to allow a full charge once a month (basically, the battery cells you do not use, die)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
If I keep the feature "limit battery charge" enabled all the time, I want the Mac to go full charge once a month anyway :)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

xsnaruto, 2023-05-07:
Same, but I prefer to get a button which allowed one time full-charge, then, I could simple click it before I go out 😄

actuallymentor, 2023-07-20:
I'm open to a PR for this, but do not have the bandwidth to implement it.

ballo, 2024-01-26:
Do you have a source for this maintenance schedule?

seefood, 2024-02-12:
this could be solved with cron in a way. there's no callback when reaching a certain level, but you could improvise this easily with a few bash lines and a semaphore file, or a log.