actuallymentor/battery #141: Error: battery: command not found while installing

ZibYang, 2023-05-08:
**What is the issue? (required)**
Got an error when launch this app at the first time. (installed through brew with `brew install battery`)

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. install with `brew install battery`
2. Click on the app icon

**Expected behavior (required)**
The app successfully lanunched

**Screenshots (optional)**
[image]

**Error logs**
Error installing battery limiter:
Command failed: PATH=/bin:/usri bin:/usr/local/bin:/usr/sbin:/opt/ homebrew battery maintain recover
/bin/bash: battery: command not found

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.
part of the log:
```
....
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain recover
Install result success

App initialisation process complete
Update/install error:
Error: Command failed: PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain recover
/bin/bash: battery: command not found
```
**Additional context (optional)**

taowaaCool, 2023-05-09:
I meet the same problem.

dotSpecs, 2023-05-10:
I tried again and solved this problem.

iSilentP, 2023-05-14:
I am also facing same problem. Tried twice to install. Both times from dmg file.

ZhiquanW, 2023-05-16:
I solved this problem by installing the CLI tool separately.

ZibYang, 2023-05-22:
I solved this problem by clearing the log file and re-installing it as before, so I will close this issue. Thanks for everyone's help :D

closed 2023-05-22 by ZibYang