actuallymentor/battery #145: battery update not updating to the latest (v1.1.4) version

Nick-Z0, 2023-06-07:
**What is the issue? (required)**
`battery update` not updating to the latest version.

I see that the latest version is 1.1.4, but when running the `battery update` command I get this message:
```
> battery update
No need to update, offline version number v1.0.11 matches remote version number
```

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. In terminal just run `battery update` using the CLI version (v1.0.11).

**Expected behavior (required)**
Updating to version v1.1.4

**Error logs**
```
06/07/23-18:35:22 - 🔼🪫 Disabling battery discharging
06/07/23-18:35:44 - Battery at 83% (target 70%)
06/07/23-18:36:44 - Battery at 84% (target 70%)
06/07/23-18:37:22 - Charge above 85
06/07/23-18:37:22 - 🔌🪫 Disabling battery charging
06/07/23-18:37:44 - Battery at 85% (target 70%)
06/07/23-18:38:44 - Battery at 85% (target 70%)
06/07/23-18:39:44 - Battery at 85% (target 70%)
06/07/23-18:40:44 - Battery at 85% (target 70%)
06/07/23-18:41:44 - Battery at 85% (target 70%)
06/07/23-18:42:44 - Battery at 85% (target 70%)
06/07/23-18:43:44 - Battery at 85% (target 70%)
06/07/23-18:44:44 - Battery at 85% (target 70%)

🖥️  Battery GUI logs:

tail: /Users/User/.battery/gui.log: No such file or directory

📁 Config folder details:

total 776
drwxr-xr-x   3 User  staff    96B Jun  7 18:35 .
drwxr-xr-x@ 85 User  staff   2.7K Jun  7 18:45 ..
-rwxr-xr-x   1 User  staff   351K Jun  7 18:45 battery.log

⚙️  Battery data:

06/07/23-18:45:08 - Battery at 85% (attached; remaining), smc charging disabled
Battery CLI utility v1.0.11
```

PS: I am trying to fix the working of the app since the version I am using does not seem to work as can be seen from the logs (target 70% but battery still charged to 85%).

Nick-Z0, 2023-06-07:
After seeing that the latest version was also available as an homebrew cask, I unistalled my existing version and installed the app from homebrew. 

Then again after checking the logs it still says `v1.0.11` even though on the GUI the latest version is shown`v1.1.4`. Is there a different version number for the CLI and GUI version?

actuallymentor, 2023-07-20:
`battery update` is for updating the CLI, not the frontend app :)

closed 2023-07-20 by actuallymentor