actuallymentor/battery #147: `battery maintain` doesn't do discharging.

parsanoori, 2023-06-16:
**What is the issue? (required)**
`battery maintain` doesn't do discharging.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Install the CLI version
2. When battery is above 80
3. Run `battery maintain 80`
4. Discharge is not enabled. As logs show.

**Expected behavior (required)**
Discharge being enabled.

**Error logs**
```
06/16/23-15:13:26 - Not triggering discharge as it is not requested
06/16/23-15:13:26 - Charging to and maintaining at 80% from 84%
06/16/23-15:13:26 - Charge above 80
06/16/23-15:13:26 - 🔌🪫 Disabling battery charging
```

lilianmoraru, 2023-06-22:
@parsanoori You have to enable the "Force Discharge" option.
It seems like it was the default behaviour but users complained about it, so now it's an option you need to activate.

actuallymentor, 2023-07-20:
Thanks for clarifying @lilianmoraru

closed 2023-07-20 by actuallymentor