actuallymentor/battery #149: Starting the battery indicator resets the limit to 80%/ignores user settings

lilianmoraru, 2023-06-22:
**What is the issue? (required)**
I use at home `battery maintain 70` to limit the charging to 70% but I've observed that the "Battery" app resets the charging limit to 80% from time to time without input from the user.
100% reproducible when the indicator is closed(`Quit` option) and then opened again.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Open `battery` app/indicator
2. From `Terminal`, execute: `battery maintain 70`
3. Check the `battery` indicator for the change to 70%, after which you `Quit` the app
4. Open the `battery` app/indicator again and see that it ignores user settings and sets the limit back to 80%.

**Expected behavior (required)**
The `battery` indicator should not reset the user settings, it should query the current limits and display that value instead, without changing it.
If no value was set, it should set it to 80%(first boot/init).

Kriepke, 2023-06-24:
indeed, the same

actuallymentor, 2023-07-20:
Can you confirm this is still an active problem in 1.1.5?

wpiekutowski, 2023-07-21:
I confirm this is still happening in 1.1.5.

armoman92, 2023-07-24:
Yeah.  Confirmed on v1.1.5.
-----
Also, the GUI does update its info to the value set in /Users/XXXX/.battery/maintain.percentage. 

 It's just static at "... 80%".  The forced discharge text is also static at "... 80%"

mrzhjyw, 2023-07-26:
The latest version has the same problem

earthsound, 2023-07-27:
In the gui.log, I'm seeing an indication that after checking for an update, the following process is run 
```Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain 80``` 
shortly after app initialization.

Just before this, maintain was set to 75.

Snippet:
```Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Found 3 battery related processed to kill
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain stop
Executing pkill -f "/usr/local/bin/battery.*"
Updating battery...   
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery update silent
Update result:
No need to update, offline version number v1.0.12 matches remote version number

Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain recover
App initialisation process complete
Executing nohup curl "https://unidentifiedanalytics.web.app/touch/?namespace=battery" > /dev/null 2>&1
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
Force discharge setting: undefined undefined
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain 80
Battery status:
{"percentage":"18","remaining":"2:24","charging":true,"discharging":false,"battery_state":"18% (2:24 remaining)","daemon_state":"smc charging enabled"}
enable_battery_limiter exec complete
Refreshing tray icon...
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv  
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status
Battery status:
{"percentage":"18","remaining":"2:24","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"18% (2:24 remaining)","daemon_state":"smc charging enabled"}
Limiter status message:
07/26/23-18:53:22 - Battery at 18% (2:24 remaining), smc charging enabled
07/26/23-18:53:22 - Your battery is currently being maintained at 80%```

earthsound, 2023-07-27:
Is this related to #128?

lilianmoraru, 2023-07-27:
@earthsound Yes, it seems to be the same issue. I think I searched for "reset" and "settings" and did not find a matching issue.

PR #166 (2023-08-20): CLI GUI compatibility
Fix for issue #149 and #128: The percentage set in the CLI is not persistent upon GUI reboot, and reset to 80%. 

**Problem**: On launch of the GUI, the maintain.percentage file is deleted, causing the GUI to recreate the file with default value of 80.
this is caused by the call "battery maintain stop" which deletes the maintain.percentage file. 

This issue is easily reproduced by changing the `battery maintain 70` via CLI, and then launching the GUI. 
 
**Solution**: Remove the deletion of file maintain.percentage in "battery maintain stop" to keep persistence of any pr [... cut]
closed 2023-08-20 by actuallymentor