actuallymentor/battery #150: Sailing Mode like in AlDente

Kriepke, 2023-06-23:
Is it possible to set, for example, to stop charging at 80 percent and turn on only below 45 percent?

IIodyne, 2023-07-16:
I don't know exactly how AlDente does it but I would reccomend making a shortcut to do it for you. Here's one I threw together:
https://www.icloud.com/shortcuts/49872c533bae4316a106851fd609d2fc

I chose a range between 40% and 60%. At the end you set how often it should run the script (currently set for 5x60 seconds = 5 minutes). At the beginning I have it set to repeat 12x24x30 times = 12 5-minutes / hours x 24 hours/day x 30days. Obviously you can change the times around. 

Let me know if you come up with a better implementation; good luck!

xoralex, 2023-07-18:
> 

If I understand correctly for the sailing mode it is better to use "battery charging off" and "battery charging on" instead of "battery discharge 40" and "battery charge 60".

Kriepke, 2023-07-19:
Thank You, I will try.

> I don't know exactly how AlDente does it but I would reccomend making a shortcut to do it for you. Here's one I threw together: https://www.icloud.com/shortcuts/49872c533bae4316a106851fd609d2fc
>
> I [...]

actuallymentor, 2023-07-20:
This is a scripting use and beyond the intended scope of this app. That said, I am open to a PR if you really care about the feature.

jakjakob, 2023-07-20:
> I don't know exactly how AlDente does it but I would reccomend making a shortcut to do it for you. Here's one I threw together: https://www.icloud.com/shortcuts/49872c533bae4316a106851fd609d2fc
>
> I [...]

Don't know if it works very good, on iPhone shortcuts eventually stop (even while running)
AND additionally, running an itself repeating shortcut isn't very efficient and is power and CPU intense. (Also just having the shortcuts app open isn't very efficient...)
(Tried it with a simple loop like you did with only a one sec. wait action on a M2 MacBook Air)

Kriepke, 2023-07-23:
While the charging of the MacBook is paused, i.e. the charge level is exactly at the set charge limit, the power supply is primarily used as a power source. However, small discharges of the battery still occur when a lot of power is needed for a short time. Then the battery of the MacBook steps in as a buffer. If the charge level drops below the charge limit due to these short discharges, Battery-app recharges the battery to the desired charge limit. A second effect on why the battery loses charge is time. If a battery is left for weeks without charging or discharging, it will also lose charge over time. Also, in this case, Battery app would detect that the charge level has dropped below the charge limit and would recharge the battery to the selected charge limit. Furthermore, if you unplug your MacBook only for a short time and use only a couple of percent, your Battery-app would charge MacBook again. Since it is theoretically healthier for the battery to be charged 10% once instead of 1% 10 times.

landabaso, 2023-08-16:
Similarly, for those that prefer using cron, add this file in `$HOME/.local/bin/batterySailingMode.bash`:

```bash
#!/bin/bash

# Fetch the battery percentage
BATTERY_PERCENTAGE=$(battery status | awk -F' ' '{print $5}' | tr -d '%')

# Compare the battery level and adjust charging
if (( BATTERY_PERCENTAGE > 60 )); then
    battery charging off
elif (( BATTERY_PERCENTAGE < 40 )); then
    battery charging on
fi
```

Then `chmod +x $HOME/.local/bin/batterySailingMode.bash`
Then `crontab -e`, which will open your favourite editor and add this line to run the script every 5 minutes:

`*/5 * * * * /Users/<YOUR USERNAME>/.local/bin/batterySailingMode.bash`

yourivdlans, 2023-09-04:
Thanks @landabaso! This is perfect.

The `BATTERY_PERCENTAGE` also contained a newline and "is" string for me, changing this line to the following worked for me:

```sh
BATTERY_PERCENTAGE=$(/usr/local/bin/battery status | awk -F' ' '{print $5; exit}' | tr -d '%')
```

dJani97, 2023-12-10:
This would be great, the goal of this program is to _prolong the longevity of the battery_, so in theory this should help in achieving that goal.

Using cron jobs and shortcuts is inconvenient because when I actually want to charge my Macbook to a 100%, these automations will keep setting the charge limit every few minutes.

Kaiman77, 2024-02-18:
I absolutely agree - at the moment it can happen, that you use the macbook withour power supply for 10 minutes, bringing battery state to 78% and it will charge it immediately back to 80....not necessarily helpful for the batter lifetime.

Usually what you use for such situations is a hysteresis with two limits.

antonurankar-moloco, 2024-12-23:
Micro-charging does not damage Li-ion batteries—in fact, it has notable benefits. One key advantage is that it helps maintain lower battery temperatures. For instance, charging a battery from 77% to 80% generates significantly less heat compared to charging it from 50% to 80%. This reduced thermal stress can extend the overall lifespan of the battery.