actuallymentor/battery #152: Can we have an option to turn on/off low-power mode

victorwon, 2023-06-29:
**Is your feature request related to a problem? Please describe.**
It'd be of great  convenience to have a menu option to turn on/off low-power mode, it's especially useful for m2 chips and the system settings are buried too deep to use. It'd be even better to have a keyboard short-cut associated with it.

**Describe the solution you'd like**
cmd to get status: pmset -g|grep "lowpowermode"|tail -c 2
cmd to set low-power mode: sudo pmset -a lowpowermode 1
cmd to unset:  sudo pmset -a lowpowermode 0

actuallymentor, 2023-07-20:
What is the advantage of doing this? I'm not familiar with this setting.

actuallymentor, 2023-07-20:
Duplicate of #110, please continue discussion there

closed 2023-07-20 by actuallymentor