actuallymentor/battery #162: Battery stops at 80%

mihainigrini, 2023-08-10:
Installed but ...
Steps to reproduce the behavior:
1. Clicked on enable battery limit
but
Battery charging stops at 81%? - MacBookPro M2 Max stock any ideea why it does not stop 80%.? Could other or previous apps have done something to cause this behaviour ?

[image]

mihainigrini, 2023-08-11:
So, after doing what I mentioned, the battery indicator remains at 80%

[image]
Just remember to disable in Battery the Optimised Battery Charging setting
[image]

mihainigrini, 2023-08-11:
If it goes to sleep, yet again it goes up  to 81 %
[image]

pdcastro, 2023-08-19:
> If it goes to sleep, yet again it goes up to 81 %

To be fair, for some time I was using the popular [AlDente app](https://apphousekitchen.com) on an M2 MacBook Air and AlDente also had this issue where the battery charged a bit beyond the configured limit, usually between 1% and 5% above the limit, especially while the laptop was sleeping while plugged in. I suspect it is _not_ some simple logic bug in AlDente or in this `battery` tool, but rather a case of the Mac itself not quite always doing what the [SMC](https://en.wikipedia.org/wiki/System_Management_Controller) settings are meant to do. Another potential contributing factor — I guess — is that between going to sleep and waking up 10 hours later, the battery voltage changes a bit due to changes in temperature and chemistry and macOS re-evaluates what it translates to in terms of state of charge. The physical world can be messy, right? :-) Anyway, just a guess.

My workaround is to set the limit a few percentage points below the desired state of charge. For example, I used to set AlDente to 75% so that typically the battery charge would reach 77-79%.

closed 2023-08-24 by mihainigrini