actuallymentor/battery #168: Running behing a proxy

lfoppiano, 2023-08-17:
**What is the issue? (required)**

I'm behind a proxy and when I started "battery" I got the following: 
[image]
[image]

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior (required)**
A clear and concise description of what you expected to happen.

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem. On a Mac you can click `control`+`command`+`shift`+`4` to create a screenshot that you can paste into this field with `command`+`4`.

**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.

**Additional context (optional)**
Add any other context about the problem here.

earthsound, 2023-08-19:
> **Error logs** I cannot open them

Runing `battery logs` may provide useful logs to share

actuallymentor, 2023-08-20:
Hey!

First fun clones the Github repo in the background, not sure how your network is set up but I presume that it blocks the app from talking to Github somehow.

Updates will not work, these require internet too :)

closed 2023-08-20 by actuallymentor