actuallymentor/battery #171: Error installing battery limiter: undefined

Zukushimeku, 2023-08-28:
**Error installing battery limiter: undefined**
[image]

When I open battery.app this error appears

The main reason is probably `Update/install error:  /usr/local/bin/battery: line 588: /Users/zukushimeku/Library/LaunchAgents/battery.plist: Permission denied` but I don't know how can I fix this.

I tried to find `battery.plist` but `/Users/zukushimeku/Library/LaunchAgents` was empty.

Here is termial records when I try to open it from `/Applications/battery.app/Contents/MacOS`

```
Last login: Mon Aug 28 23:00:53 on ttys000
/Applications/battery.app/Contents/MacOS/battery ; exit;
zukushimeku@MacBook ~ % /Applications/battery.app/Contents/MacOS/battery ; exit;
[ update-electron-app ]  feedURL https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.0
[ update-electron-app ]  requestHeaders { 'User-Agent': 'update-electron-app/2.0.1 (darwin: arm64)' }
Starting tray app
Get active logo for 100
Display percentage 100 based on 100
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Tray app boot complete
Triggering boot-time auto-update
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://icanhazip.com &> /dev/null
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://github.com &> /dev/null
[ update-electron-app ]  checking-for-update
[ update-electron-app ]  update-not-available
Internet online: true
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which battery
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which smc
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0I -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k ACLC -r
Is installed?  /usr/local/bin/smc

Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Found 2 battery related processed to kill
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain stop
Executing pkill -f "/usr/local/bin/battery.*"
Error killing existing battery progesses, usually means no running processes
Updating battery...
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery update silent
Update result:  No need to update, offline version number v1.1.5 matches remote version number

Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain recover
Update/install error:  /usr/local/bin/battery: line 588: /Users/zukushimeku/Library/LaunchAgents/battery.plist: Permission denied

App initialisation process complete
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

``` 

I think this is not necessary but I am using Macbook Air 2020 M1.

llamahunter, 2023-09-02:
This seems to happen when I try to launch the app with the finder. But if I run it from the command line, it works?

llamahunter, 2023-09-02:
It turns out I was executing an alias in my .zshrc, and for some reason, while that works from the command line, it does _not_ work when launching 'battery' via the Finder.  Rewriting my .zshrc to not do that fixed this.

rivolity, 2023-09-13:
@llamahunter @Zukushimeku 

You can fix that by following these steps.
*I had the same probleme yesterday*
$\color{green}{Step:1}$
- Ensure that the Battery is installed, even if you have the following error message ` Error installing battery limiter: undefined ` just don't uninstall it.
Run this command in terminal
```
which battery
```
You should get this output
```
/usr/local/bin/battery
```
$\color{orange}{Step:2}$
- Run the following command to update the Battery app from the command line.
*this will call the shell script and download the latest script from the git repo and execute the installation via bash.*

```
battery reinstall
```
Press any key to continue and hit enter, this will prompt you to enter the root password, since the shell script need root access to visudo and other privilege.

Here is a screen capture of the running command

[image]

Here is the latest version after the update via the CLI.

[image]

Zukushimeku, 2023-09-13:
@rivolity Thanks for respond but it didn't work for me. I tried so many times reinstalling with all ways but it didn't work at all. After reinsalling again it seems like nothing happen, same terminal screen but still not opening. 
So I tried using `battery visudo`command but it also didn't work and responded this 
[image]
I don't know why but it says you don't have permission every time...

actuallymentor, 2024-01-17:
Can you let me know if the latest release works for you? Feel free to re-open if you have follow-up questions.

closed 2024-01-17 by actuallymentor

dchakro, 2025-02-15:
I'm in the same situation as @s-gbz and @Zukushimeku.
I did however tried to run the GUI from the terminal to try & capture what's going on, and seems that there's a `grep` that is causing the error on my end.

`Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Update/install error:  grep: warning: stray \ before d`

**Complete log:**
```sh
~ /Applications/battery.app/Contents/MacOS/battery
[ update-electron-app ]  feedURL https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.2
[ update-electron-app ]  requestHeaders { 'User-Agent': 'update-electron-app/2.0.1 (darwin: arm64)' }
Starting tray app
Get active logo for 100
Display percentage 100 based on 100
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Tray app boot complete
Triggering boot-time auto-update
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin curl -I https://icanhazip.com &> /dev/null
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin curl -I https://github.com &> /dev/null
[ update-electron-app ]  checking-for-update
[ update-electron-app ]  updater error
[ update-electron-app ]  [Error: bad URL] { code: -1000, domain: 'NSURLErrorDomain' }
Internet online: true
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which battery
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which smc
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0I -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -w 02
Is installed?  /usr/local/bin/smc

Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Update/install error:  grep: warning: stray \ before d

App initialisation process complete
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
```
P.S. I tried reinstalling with `battery reinstall` and other methods listed by people several times.
[image]

s-gbz, 2025-02-18:
@dchakro - I have founded another solution that works for me - installing "AlDente" (https://apphousekitchen.com) - it's not OSS, but it's free and get's the job done perfectly.