actuallymentor/battery #172: Mag safe led new feature not working in some scenarios

Arad-Afzali, 2023-08-31:
I tried the new update that lets you change the color of the magsafe when the battery is 80% full, but it didn't work well for me.

It was fine right after I restarted my mac, but it got messed up after sleep mode. The problem was that the led light never turned green again.

I tried this several times over a few days and the result is always the same.

By the way, I think it might have some issues with functioning when "force discharge" is on, but I'm not sure.

Arad-Afzali, 2023-09-02:
Hey, I just tested it again and this is what I found out:
The led stays green until I unplug and plug the charger back into the macbook. Then it turns red and never changes back to green.

jd1900, 2023-10-04:
I'm not having green either.
I connect when already at 80%. Never turns green.

Fizzadar, 2023-11-23:
This is because the maintain script only applies changes when the battery level crosses the threshold.  https://github.com/actuallymentor/battery/pull/20 should fix it.

actuallymentor, 2024-01-17:
Should be fixed in the latest release!