actuallymentor/battery #174: Surfing Mode: Maintain a charge RANGE instead of value

dempstert, 2023-09-04:
Instead of maintaining a fixed charge level, eg 80%, can we have the feature to maintain a charge level between a lower and upper threshold?

Maintaing a fixed percentage means a lot of switching between charging / discharging mode.

I assume it's better for battery health in the long term to run between 50% and 90% smoothly, than to sit and 80% and constantly modulate.

pfitoussi, 2023-09-05:
It seems that you can apply this feature with :
```
battery charge 90
battery discharge 50
```

my4ng, 2023-09-06:
> battery charge 90
battery discharge 50

these are only one-off commands, cf. `battery maintain`.

dempstert, 2023-09-07:
@my4ng Does `battery maintain` allow settings for upper & lower windows ?

my4ng, 2023-09-07:
@dempstert I do not believe so. I haven't read through the source code, so I can only suspect that it is a binary operation (i.e. `< threshold`) rather than a ranged one. On the other hand, Al Dente does support the so-called "sailing mode", which means this is not a hardware limitation either.

earthsound, 2023-10-24:
Dupe of #150

actuallymentor, 2024-01-17:
Closing as duplicate of #222, let's continue the convo there

closed 2024-01-17 by actuallymentor