actuallymentor/battery #176: FR: Better and more useful GUI

oliver408i, 2023-09-07:
**Is your feature request related to a problem? Please describe.**
I wanted this app mostly because I want to set a custom charge limit as well as enable and disable battery charging whenever I wanted. However, the GUI app that comes with this is very limited and is very useless to me. The CLI got me covered, but I wanted a GUI for it and I think others like me who can't code will find it useful too.

**Describe the solution you'd like**
I actually made an app that acts like a GUI wrapper for this: https://github.com/oliver408i/mac-battery-controller
(I'm still working on it, testing it, and fixing it as I go on)

If you want to just take my app and just post it here, I'm fine with that. If you want to make your own app, I'm also fine with tha.

**Describe alternatives you've considered**
[See the app I made]
I already tried aldente but the free licence is basically useless.

**Additional context**
n/a

jakjakob, 2023-09-18:
this application is meant to be a minimalistic (and lightweight) application, a richer GUI isn't planned. Maybe some more options in the existing GUI

oliver408i, 2023-09-18:
Yes it is meant to be minimalistic, but at least it should have all the features that the cli version offers

actuallymentor, 2024-01-17:
The GUI will remain much more limited than the CLI I'm afraid. But I welcome your efforts for an alternative GUI!

When yo consider it mature, I am happy linking to it from the `README.md` :)

closed 2024-01-17 by actuallymentor