actuallymentor/battery #178: M2 Air charges more than 80%

hardcod3dd, 2023-09-11:
**What is the issue? (required)**

M2 Air charges more than 80%

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
My laptop was open.software was running as usual.battery at 80%,then I shutdown the computer.remove the magsafe and after 1-2 hours open up the laptop again and the battery is at 83%.

**Expected behavior (required)**
battery should be 80 or less

**Screenshots (optional)**

**Error logs**

jakjakob, 2023-09-18:
probably  https://github.com/actuallymentor/battery/pull/163 will fix the problem (for 80% charge limit)

hardcod3dd, 2023-09-19:
its interesting to see apple implemented a firmware based limit.would be very useful for it to work even when the laptop is powered off.thanks for the tip.closing the issue for now.

closed 2023-09-19 by hardcod3dd