actuallymentor/battery #179: Clamshell force-discharge

dennyblokland, 2023-09-13:
_**Is your feature request related to a problem? Please describe.**_
It's a bit of a shame that [force-discharging] doesn't work in clamshell mode. Al Dente seems to have this ability in the pro package. However, I like this app way more for its elegance (simplicity) and price. 

_**Describe the solution you'd like**_
Add the ability to force-discharge in clamshell mode. 

_**Describe alternatives you've considered**_
For now I'm keeping the lid half-open to allow force-discharge. (I haven't yet got to the point where I find out if I can close the lid after the 80% is reached, and keep it there)

_**Additional context**_
Add any other context or screenshots about the feature request here.

Thanks!!

actuallymentor, 2024-01-17:
Pull requests are welcome for this!

42648594, 2024-03-07:
Ditto on this. It seems AlDente, this app, and the other apps designed to take control of Optimized Charging all break in clamshell mode. I wonder if simulating the state of clamshell to macOS or the app would be of any use?

0x2CA, 2024-04-16:
> Ditto on this. It seems AlDente, this app, and the other apps designed to take control of Optimized Charging all break in clamshell mode. I wonder if simulating the state of clamshell to macOS or the app would be of any use? The AlDente dev said it's "a hardware limitation" which seems hard to believe given Optimized Charging itself can activate in clamshell mode, though of course that's of little use to anyone given how sporadic and unreliable it is.

https://apphousekitchen.com

[image]