actuallymentor/battery #182: Error installing battery limiter: Command failed: PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery maintain recover /bin/bash: battery: command not found

kylin-kai, 2023-09-15:
[image]

flyingCrp, 2023-09-23:
+1  M1 Pro

yuyedaidao, 2023-09-23:
+2 M2 Pro

yujajoe, 2023-09-25:
+1 M2 Max macOS 14.0

xxxbetterrun, 2023-09-26:
+2 M2 Pro

seyoulala, 2023-09-26:
+1 M1 PRO

cratanda, 2023-10-07:
+1 M2 air

Haileysu, 2023-10-09:
+1,M1 pro

wookayin, 2023-10-09:
Dear community, please don't post **spammy** "me, too" comments. It's not going to be helpful.

This error message would happen when the battery CLI has failed to install. If you encounter this error, please install the CLI manually from terminal (https://github.com/actuallymentor/battery#installation-1) or restart the app to start the auto-installer again. It will also tell you, if something goes wrong, why the installer has failed (BTW a very common reason is entering a wrong password).

You can verify that the `battery` CLI is installed well:
```
$ which battery
/usr/local/bin/battery
```

Once this is ensured, the error will be gone.

(NOTE: Somehow the installer puts the binary into `/usr/local/bin`, not `/opt/homebrew/bin`... 
 though in the future the `battery` CLI can be automatically installed as a part of homebrew installation)

flyingCrp, 2023-10-11:
> Dear community, please DO NOT **spammy** "me, too" comments. It's not going to be helpful.
> 
> This error message would happen when the battery CLI has failed to install. If you encounter this error, please install the CLI manually from terminal (https://github.com/actuallymentor/battery#installation-1) or restart the app to start the auto-installer again. It will also tell you, if something goes wrong, why the installer has failed (BTW a very common reason is entering a wrong password).
> 
> You can verify that the `battery` CLI is installed well:
> 
> ```
> $ which battery
> /usr/local/bin/battery
> ```
> 
> Once this is ensured, the error will be gone.
> 
> (NOTE: Somehow the installer puts the binary into `/usr/local/bin`, not `/opt/homebrew/bin`... though in the future the `battery` CLI can be automatically installed as a part of homebrew installation, [like smc](https://github.com/actuallymentor/battery/tree/main/dist))
> 
> BTW, the installer script could be improved so that it can better detect and handle a situation where [installation of the battery CLI has failed](https://github.com/actuallymentor/battery/blob/main/app/modules/battery.js#L162-L174).

thank you for reply. i fixed this with your reply

mcaif, 2023-11-16:
I had the same problem but found the cause.

My ISP was blocking the zip url from code load.github.com where's located. I used a vpn and the script ran smoothly.

Ashu-r, 2023-11-20:
> I had the same problem but found the cause.
> 
> My ISP was blocking the zip's url from codeload.github.com where's located. I used a vpn and the script ran smoothly.

Thanks @mcaif I had similar problem. Using Cloudflare warp worked for me.

actuallymentor, 2024-01-17:
Oh very interesting, I had not considered that ISPs might block github. Do you mind sharing the country you are in?

I'm not entirely sure how to fix this issue, other than to give out an error perhaps when Github cannot be reached.

mcaif, 2024-01-21:
@actuallymentor I'm from Spain. My ISP is Vodafone.

I can't think a solution to evade this type of blocking. Maybe commit the entire zip on the repo? The error message should be enough.

Blackspell01, 2024-02-27:
I don't understand what to do to solve this problem. I tried all tree installation methods. Always fails. Typing "battery help" returns "zsh: command not found: battery"

Barry1915, 2024-08-22:
> Dear community, please DO NOT **spammy** "me, too" comments. It's not going to be helpful.
>
> This error message would happen when the battery CLI has failed to install. If you encounter this error, [...]

thanks!!!!