actuallymentor/battery #186: Mac OS sonoma

anarchy89, 2023-09-27:
Before I upgrade to sonoma, are there any issues using this with sonoma?

hardcod3dd, 2023-09-27:
im using it with Sonoma without any issue

anarchy89, 2023-09-27:
Thanks, this is the only app I really need to work haha

Mitcal, 2023-10-11:
I've upgraded to Sonoma and just noticed today that the battery has been charged to 100%. Was only charging to 80% before the upgrade. I've re-installed the app now and started a forced-discharge. We'll see what happens.

anarchy89, 2023-10-11:
Hmm, it works fine for me. LED stays on orange when I run battery charging off. Didn't need to reinstall or upgrade it. Maybe there's an issue with the newer version?

nicolas-albert, 2023-10-12:
@Mitcal the charging 100% is normal because during the update process, no other software run even for the battery limit.

For the led, yesterday it turned green at 80% as expected.