actuallymentor/battery #188: Opt out of telemetry or usage statistics data

wookayin, 2023-10-09:
First of all, thanks for the great app.

https://github.com/actuallymentor/battery/blob/main/app/modules/battery.js#L177

This app will collect user data or usage statistics without any constent, and I would like to opt out of it. There should at least some kind of notice to users informing that some kind of data is being collected.

anarchy89, 2023-11-03:
Is anything being done about this?

actuallymentor, 2024-01-17:
Thanks for the request @wookayin!

The data collected is simply how many times it is opened and from where, no behavioural tracking there. It's just for me to know whether people are using the app and it's worth my time to keep working on it.

If you care deeply about this, I don't mind a pull request that adds it under "advanced settings" of the GUI.