actuallymentor/battery #193: battery won't start, could not enable service: 125

leakitisrossi, 2023-11-06:
**What is the issue? (required)**
battery won't start

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Open battery, both GUI or CLI

**Expected behavior (required)**
Application to start

**Error logs**
sudo battery maintain 80
11/06/23-08:32:26 - Starting battery maintenance at 80%
11/06/23-08:32:26 - Writing new setting 80 to /Users/user/.battery/maintain.percentage
11/06/23-08:32:26 - Maintaining battery at 80%
11/06/23-08:32:26 - Daemon already exists, checking for differences
Could not enable service: 125: Domain does not support specified action

`tail -n 1000 ~/.battery/*/log` gives "no matches found", and `battery log` gives no output

actuallymentor, 2024-01-17:
Very weird, have you tried reinstalling?

matteoipri, 2025-09-22:
Hi! I have the same issue, even after reinstalling.
```
$ sudo battery reinstall
This will run curl -sS https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash
Press any key to continue

####################################################################
# 👋 Welcome, this is the setup script for the battery CLI tool.
# Note: this script will ask for your password once or multiple times.
####################################################################

🔋 Starting battery installation
[ 1 ] Superuser permissions acquired.
[ 2 ] Downloading latest version of battery CLI
[ 3 ] Move smc to executable folder
[ 4 ] Writing script to /usr/local/bin/battery for user root
[ 5 ] Setting correct file permissions for root
[ 6 ] Setting up visudo declarations
09/22/25-08:34:07 - Setting visudo file permissions to root
The existing battery visudo file is what it should be for version v1.2.7
[ 7 ] Removing temp folder /Users/matteo.ipri/.battery-tmp

🎉 Battery tool installed. Type "battery help" for instructions.

$ sudo battery maintain 70 --force-discharge
09/22/25-08:35:29 - Killing old maintain process at 67195
09/22/25-08:35:29 - Starting battery maintenance at 70% --force-discharge
09/22/25-08:35:29 - Writing new setting 70 to /Users/matteo.ipri/.battery/maintain.percentage
09/22/25-08:35:29 - Maintaining battery at 70%
09/22/25-08:35:29 - Daemon already exists, checking for differences
Could not enable service: 125: Domain does not support specified action
```

I am running macOS Sonoma 14.8 (23J21). After last update to 14.8 the battery is not kept to 70% max as expected, that is what triggered a reinstall for me and seeing again the error message.