actuallymentor/battery #194: Add sailing mode like Al dente. Also the magsafe support, when battery is at set battery level, it will turn green and stop charging.

Confidack, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

panzer-arc, 2023-11-22:
My MagSafe light is green and charging is limited to 80%. I think that may already be included.

sanketshahanebookendai, 2023-12-01:
I think what @Confidack is asking for is to let the battery fluctuate between say 70% and 80% in order to avoid trickle charging.

actuallymentor, 2024-01-17:
Issue template is not decorative.

closed 2024-01-17 by actuallymentor