actuallymentor/battery #197: Uninstall?

seamusdemora, 2023-11-18:
Pardon me for asking what's probably a dumb question, but after a couple of unpleasant experiences a few years ago, I like to always ask this question before installing anything: 

**_How do I un-install this app?_** 

**What is the issue? (required)**
A clear and concise description of what the bug is.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior (required)**
A clear and concise description of what you expected to happen.

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem. On a Mac you can click `control`+`command`+`shift`+`4` to create a screenshot that you can paste into this field with `command`+`4`.

**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.

**Additional context (optional)**
Add any other context about the problem here.

hexclann, 2023-11-23:
Using `battery uninstall` command in the terminal

seamusdemora, 2023-11-24:
uh, Yeah - I noticed that after posting this... my apologies.

closed 2023-11-24 by seamusdemora