actuallymentor/battery #203: Battery limiter error: Cannot read properties of undefined (reading 'includes')

rosch100, 2023-11-25:
**What is the issue? (required)**
An error message displaying 
'Battery limiter error: Cannot read properties of undefined (reading
'includes')'
popped up. The battery icon was great, but after clicking on it, all info was displayed correctly.

I could not find any log entries in the system console. The battery log did not record any problems.
[battery.log](https://github.com/actuallymentor/battery/files/13463520/battery.log)
[image]

benzht, 2023-11-27:
Same here. And the logs show
🚨 Missing image: /Applications/battery.app/Contents/Resources/battery-active-NaN-Template.png

dynek, 2023-12-11:
I'm having the same popup here on a M1 Pro.

When this occurs, is it also freezing your GUI ?

maxemillianian, 2023-12-31:
M2 MacBook Air: When resuming from sleep, I'm getting the popup. No system freezes, though.

actuallymentor, 2024-01-17:
Could you try the latest release and re-open this issue if the problem persists?

closed 2024-01-17 by actuallymentor