actuallymentor/battery #208: It would be nice if your application would respect XDG_CONFIG_HOME ($HOME/.config)

rafo, 2023-12-05:
**Is your feature request related to a problem? Please describe.**
It would be nice if your application would respect the [XDG_CONFIG_HOME](https://wiki.archlinux.org/title/XDG_Base_Directory), which defaults to $HOME/.config. It's nice to have an uncluttered $HOME!

**Describe the solution you'd like**
It looks like **setup.sh** defines some temp- and configfolder settings and **battery.sh** uses them *and* start the app (?) but I didn't test it.

I dont know anything about electron apps but i think it would be much cleaner to use a standard library to read the XDG_CONFIG_HOME env. If you want, define your own env BATTERY_CONFIG etc.

**Additional context**

[https://wiki.archlinux.org/title/XDG_Base_Directory](https://wiki.archlinux.org/title/XDG_Base_Directory)

actuallymentor, 2024-01-17:
I like your thinking @rafo, an uncluttered home is nice indeed.

How is `XDG_RUNTIME_DIR` read? I get an empty state when I access it through CLI.

Do you have the bandwidth to PR for this?

twio142, 2024-07-31:
I'd also like to see `~/.battery` be moved to `$XDG_CONFIG_HOME/`.
I don't have `$XDG_RUNTIME_DIR` set neither, but `$XDG_CONFIG_HOME` would do. And `~/.battery` could be a fallback solution.

kfernandez31, 2024-09-29:
Hi @actuallymentor, do you foresee any updates on this topic?

actuallymentor, 2024-09-30:
> Hi @actuallymentor, do you foresee any updates with this topic?

Apologies for the slow responses. The time I can dedicate to battery fluctuates depending on "real" work.

This is extremely low priority for me (has next to to user impact), however I'm happy to merge a PR for it.

Anyone making a PR: please add fallback logic and remember that the paths need to be synced between the cli code (bash) and the GUI code (js).

twio142, 2024-09-30:
@actuallymentor It's because the XDG-env var is a linux thing. For mac to use them they have to be explicitly set in the environment. Typically `XDG_CONFIG_HOME` would be set to `$HOME/.config`.

actuallymentor, 2024-09-30:
> @actuallymentor It's because the XDG-env var is a linux thing. For mac to use them they have to be explicitly set in the environment. Typically `XDG_CONFIG_HOME` would be set to `$HOME/.config`.

I'm a tad confused then... This is a Mac only app, why is the XDG variable set desirable?

twio142, 2024-09-30:
It's just about organizing all sorts of config directories you got into `~/.config` instead of having them spread in the home dir. Also it gives you consistent experience if you also use linux.