actuallymentor/battery #209: Option to Hide Battery Icon from Menu Bar for a Cleaner Interface

jcherrera, 2023-12-05:
**Is your feature request related to a problem? Please describe.**
I often find the menu bar on my device cluttered, especially with icons that I don't frequently use. The "battery" application icon is one such example. Its constant presence in the menu bar adds to the clutter without providing necessary information at a glance, as I don't often need to check my battery status.

**Describe the solution you'd like**
I would like an option in the "battery" application settings that allows users to hide its icon from the menu bar. This feature should be easily toggleable, enabling users who wish to monitor their battery status closely to keep the icon visible, while others can choose to remove it for a cleaner look.

**Describe alternatives you've considered**
An alternative could be an auto-hide feature where the icon only appears when the battery is low or in a critical state. Another possibility is integrating the battery status into another menu icon or notification center, reducing the need for a separate battery icon.

**Additional context**
This change aims to enhance user experience by reducing menu bar clutter, catering to users who prefer a minimalistic interface. The flexibility to show or hide the icon according to personal preference would be greatly appreciated.

ArnavVarshney, 2023-12-05:
Not to disregard this suggestion, but you can just "Quit" the app from the dropdown and that'll remove the menu bar icon. The battery limit is, however, still applied normally, since I think the app is just a GUI extension to a CLI backend.
However, any time you want to make a change, you would have to re-open the app.

jcherrera, 2023-12-05:
You are right, thanks! @ArnavVarshney