actuallymentor/battery #213: Possibility to pause charge when battery has reached desired %.

FUFAFUFA, 2023-12-23:
A charging mode that allows the charging to stop when it reach for example 80% (upper limit) and then when the battery reach for example 70% (lower limit), the charging will start again.

**Describe the solution you'd like**
Settings that allow me to set limits like illustrated over here.

**Describe alternatives you've considered**
None

**Additional context**
None

Cipher-08, 2023-12-24:
We can use a combination of battery APIs and a control mechanism to start and stop charging based on these limits.

actuallymentor, 2024-01-17:
Cool idea, do you have the time to draft a pull request for this?

Closing as duplicate of #222, let's continue our chat there.

closed 2024-01-17 by actuallymentor