actuallymentor/battery #214: Limit by voltage

rguca, 2023-12-26:
A lithium ion battery is optimally stored at 3.8V per cell. For the MacBook this means 11.4V for the battery.
When discharging from a full charge, 11.4V are around 40%. After one month with battery limit at 40%, the battery voltage has dropped significantly. 11.4V are around 60% now.

Currently, I adjust the percentage limit every few weeks to keep it at the desired voltage of 11.4V.
It would be very nice to integrate this limit by voltage into a tool, so that the battery state of charge stays correct, even when the percentage indicator begins to drift.

The voltage can be retrieved using:
```
ioreg -l -n AppleSmartBattery -r | grep "\"Voltage\" =" | awk '{ print $3/1000 }'
```

actuallymentor, 2024-01-17:
Ooooh this is very cool. Do you know why the % drifts when the voltage moves? I assumed the % is voltage based as in many devices.

Do you envision this as a separate command? IE `battery voltage <number>`?

rguca, 2024-01-17:
It seems like the mAh are counted when charging/discharging and the % is based on that.
This counting is not perfect and over a long period of time the mistakes accumulate and lead to the drift.
If you discharge to 0% or charge to 100% the counter is reset and the drift is gone.

A separate command would be perfect. The arguments should be an upper and lower bound.
If the voltage drops below the lower bound, charging starts. If it reaches the upper bound, it stops.
This is necessary, because the voltage will rise immediately, when charging starts and settle a bit, when ist stops.

IE `battery voltage <min> <max>`

actuallymentor, 2024-01-17:
Do you have the time to draft a pull request for this?