actuallymentor/battery #215: sudo: smc: command not found

ballo, 2023-12-27:
**What is the issue? (required)**
A clear and concise description of what the bug is.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
Run:
`battery charge 40`

**Expected behavior (required)**
Charge the battery

**Error logs**```

battery charge 40
12/27/23-11:08:47 - Killing running maintain daemons & enabling charging as default state
12/27/23-11:08:47 - Disabling daemon at gui/501/com.battery.app
12/27/23-11:08:47 - 🔌🔋 Enabling battery charging
sudo: smc: command not found
sudo: smc: command not found
12/27/23-11:08:47 - 🔼🪫 Disabling battery discharging
sudo: smc: command not found
/usr/local/bin/battery: line 152: smc: command not found
12/27/23-11:08:47 - Battery at 28% ((no remaining), smc charging disabled
12/27/23-11:08:47 - Setting adapter to on
12/27/23-11:08:47 - Killing running maintain daemons & enabling charging as default state
12/27/23-11:08:47 - Disabling daemon at gui/501/com.battery.app
12/27/23-11:08:47 - 🔌🔋 Enabling battery charging
sudo: smc: command not found
sudo: smc: command not found
12/27/23-11:08:47 - 🔼🪫 Disabling battery discharging
sudo: smc: command not found
/usr/local/bin/battery: line 152: smc: command not found
12/27/23-11:08:47 - Battery at 28% ((no remaining), smc charging disabled
12/27/23-11:08:47 - 🔽🪫 Enabling battery discharging
sudo: smc: command not found
12/27/23-11:08:47 - Charging to 40% from 28%
12/27/23-11:08:47 - 🔌🔋 Enabling battery charging
sudo: smc: command not found
sudo: smc: command not found
12/27/23-11:08:47 - 🔼🪫 Disabling battery discharging
sudo: smc: command not found
12/27/23-11:08:47 - Battery at 28%

```

**Additional context (optional)**
Add any other context about the problem here.
I installed the latest via brew. For whatever reason, this also includes the utterly useless GUI app which cannot adjust any settings.

ballo, 2023-12-27:
Wow, are you kidding me? I have to run the crappy GUI app in order for the CLI command to work. Time for a fork

marchein, 2024-01-07:
@ballo have you checked the installation guide for the cli in the readme if you don't want the gui? 
https://github.com/actuallymentor/battery#installation-1

ballo, 2024-01-09:
I'd use the GUI if it wasn't hardcoded to use settings only the author wants.

actuallymentor, 2024-01-17:
@ballo with this attitude there is a 0% probability I'm even looking into this. Go touch grass.

closed 2024-01-17 by actuallymentor

ballo, 2024-01-17:
You made it clear it's 0% in either case

earthsound, 2024-01-23:
> I'd use the GUI if it wasn't hardcoded to use settings only the author wants.

@ballo if you set the limit to e.g. 75% in the CLI the GUI will use that percentage.

ballo, 2024-01-25:
> > I'd use the GUI if it wasn't hardcoded to use settings only the author wants.
> 
> @ballo if you set the limit to e.g. 75% in the CLI the GUI will use that percentage.

Oh, thank you! I didn't know about that. 

In case anyone else is having difficulty: use the "battery maintain 40" or whatever command (not charge or discharge). Not sure if this setting is kept after a reboot, however.

earthsound, 2024-01-26:
> > > I'd use the GUI if it wasn't hardcoded to use settings only the author wants.
> > 
> > 
> > @ballo if you set the limit to e.g. 75% in the CLI the GUI will use that percentage.
> 
> Oh, thank you! I didn't know about that.
> 
> In case anyone else is having difficulty: use the "battery maintain 40" or whatever command (not charge or discharge). Not sure if this setting is kept after a reboot, however.
> 
> Edit: I'm going to have to do some testing to see if this actually works...

I use `battery maintain 75` and it maintains that level after a reboot for me.

ballo, 2024-01-26:
> I use `battery maintain 75` and it maintains that level after a reboot for me.

Unfortunately, while the battery app claims it's maintaining at a specific level, the mac itself is charging even when the battery is above that level and I plug in the power cord (based on the battery level going from below the "maintain" threshold to above it). Running battery maintain while it's plugged in doesn't stop it. Only running battery discharge stops the charging.