actuallymentor/battery #216: Error installing battery limiter: undefined

nilsml, 2024-01-02:
**What is the issue? (required)**
Trying to get the latest version of Battery running on my M2 Pro Mac, but the GUI part will not start.
CLI seems to be working, but whenever I try to start the app I gett the error message mentioned in the title.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Start battery.app on a MacOs machine

**Expected behavior (required)**
The app would start

**Error logs**
App initialisation process complete
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
[ update-electron-app ] 
update-not-available
[ update-electron-app ] 
requestHeaders
[object Object]
Tray app boot complete
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Starting tray app
[ update-electron-app ] 
feedURL
https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.0
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://github.com &> /dev/null
[ update-electron-app ] 
checking-for-update
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://icanhazip.com &> /dev/null
Triggering boot-time auto-update
Get active logo for 100
Display percentage 100 based on 100
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0I -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k ACLC -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which smc
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which battery
Internet online: false
Is installed? 
false
Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Update/install error: 
/Users/nilsml/.bash_aliases: line 19: alias: Documents/iCloud~com~logseq~logseq/Documents: not found

[ update-electron-app ] 
update-not-available
App initialisation process complete
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
[ update-electron-app ] 
feedURL
https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.0
[ update-electron-app ] 
requestHeaders
[object Object]
Tray app boot complete
Triggering boot-time auto-update
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://icanhazip.com &> /dev/null
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://github.com &> /dev/null
[ update-electron-app ] 
checking-for-update
Starting tray app
Display percentage 100 based on 100
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Get active logo for 100
Internet online: false
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0I -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k ACLC -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which smc
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which battery
Is installed? 
false
Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Update/install error: 
/Users/nilsml/.bash_aliases: line 19: alias: Documents/iCloud~com~logseq~logseq/Documents: not found

App initialisation process complete
[ update-electron-app ] 
update-not-available
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv
[ update-electron-app ] 
feedURL
https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.0
[ update-electron-app ] 
requestHeaders
[object Object]
Get active logo for 100
Display percentage 100 based on 100
Starting tray app
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Tray app boot complete
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://icanhazip.com &> /dev/null
Triggering boot-time auto-update
[ update-electron-app ] 
checking-for-update
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew curl -I https://github.com &> /dev/null
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k ACLC -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which battery
Internet online: false
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew which smc
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew sudo -n /usr/local/bin/smc -k CH0I -r
Is installed? 
false
Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Update/install error: 
/Users/nilsml/.bash_aliases: line 19: alias: Documents/iCloud~com~logseq~logseq/Documents: not found

[ update-electron-app ] 
update-not-available
App initialisation process complete
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew battery status_csv

📁 Config folder details:

total 19352
drwxr-xr-x@  6 nilsml  staff   192B Jan  2 08:33 .
drwxr-x---+ 64 nilsml  staff   2.0K Jan  2 08:39 ..
-rwxr-xr-x@  1 nilsml  staff    65K Jan  2 08:39 battery.log
-rw-r--r--@  1 nilsml  staff   8.4M Jan  2 08:35 gui.log
-rw-r--r--@  1 nilsml  staff     3B Dec 23 17:44 maintain.percentage
-rw-r--r--@  1 root    staff   897B Jan  2 08:32 visudo.tmp

⚙️  Battery data:

01/02/24-08:39:49 - Battery at 65% (attached; remaining), smc charging disabled
Battery CLI utility v1.1.5

iAsimov, 2024-01-11:
Same issue.

j0rd1smit, 2024-03-02:
@actuallymentor Can we reopen this? 
It is still an issue.
It does not matter if which of the following ways I use:
- `brew install battery` (as documented on github readme page).
- `brew install --cask battery` (as documented on homebrew page).
- the `battery-1.2.1-mac-arm64.dmg` file.

If I then open the application via Spotlight, clicking on the icon, etc, I get this error.
However, the error does not occur if open it via the terminal with `open /Applications/Battery.app`.

cfurrow, 2024-03-06:
Having the same issue:

- Apple M3 Max
- macOS 14.3.1

I opened the Console.app to see if there were any relevant logs, and saw these (I did not copy all of them, only ones that seemed relevant and were not related to macOS running board and window management):

```
error	11:58:06.108629-0500	tccd	TCCDProcess: identifier=co.palokaj.battery, pid=22948, auid=501, euid=501, binary_path=/Applications/battery.app/Contents/MacOS/battery attempted to call TCCAccessRequest for kTCCServiceAccessibility without the recommended com.apple.private.tcc.manager.check-by-audit-token entitlement

default	11:58:06.178212-0500	tccd	AUTHREQ_ATTRIBUTION: msgID=367.521, attribution={accessing={TCCDProcess: identifier=co.palokaj.battery, pid=22948, auid=501, euid=501, binary_path=/Applications/battery.app/Contents/MacOS/battery}, requesting={TCCDProcess: identifier=com.apple.WindowServer, pid=367, auid=88, euid=88, binary_path=/System/Library/PrivateFrameworks/SkyLight.framework/Versions/A/Resources/WindowServer}, },

default	11:58:06.179438-0500	tccd	AUTHREQ_SUBJECT: msgID=367.521, subject=co.palokaj.battery,

default	11:58:06.180851-0500	tccd	-[TCCDAccessIdentity staticCode]: static code for: identifier co.palokaj.battery, type: 0: 0x13a898920 at /Applications/battery.app
```

It seems there may be a permissions/entitlement issue according the first log line, which was the only real error I saw in the logs:

```
TCCDProcess: 
identifier=co.palokaj.battery, 
pid=22948, 
auid=501, 
euid=501, 
binary_path=/Applications/battery.app/Contents/MacOS/battery 

attempted to call TCCAccessRequest for kTCCServiceAccessibility 
without the recommended com.apple.private.tcc.manager.check-by-audit-token entitlement
```