actuallymentor/battery #217: Battery fails to launch on MBP M1

GHT52, 2024-01-03:
[image]
**What is the issue? (required)**
The app fails to luanch.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
Double click on the app icon

**Expected behavior (required)**
I expected it to launch

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem. On a Mac you can click `control`+`command`+`shift`+`4` to create a screenshot that you can paste into this field with `command`+`4`.

**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.

**Additional context (optional)**
Add any other context about the problem here.

krystofbe, 2024-01-05:
i have the same issue

actuallymentor, 2024-01-17:
Which version of battery did you install and where did you download it?

BrakkeTony, 2024-01-22:
Same here, with version 1.2.1, both the .dmg and the home-brew download